thonny/thonny #1272: Problems Installing in Windows 10 64Bit

Mitchell1944, 2020-07-21:
After installing thonny this is what I get when I try to run it!
OS is Windows 10 Pro Version 2004 OS Build 19041.388
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\tktextext.py", line 229, in dispatch
    return fun(event)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\shell.py", line 898, in perform_return
    EnhancedTextWithLogging.perform_return(self, event)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\tktextext.py", line 410, in perform_return
    self.insert("insert", "\n")
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\shell.py", line 860, in intercept_insert
    self._try_submit_input()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\shell.py", line 1035, in _try_submit_input
    self._submit_input(submittable_text)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\shell.py", line 1177, in _submit_input
    get_runner().send_program_input(text_to_be_submitted)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 297, in send_program_input
    self._proxy.send_program_input(data)
AttributeError: 'NoneType' object has no attribute 'send_program_input'
So if you can help me get this running I would be more than grateful.
Thanks, Mitchell

aivarannamaa, 2020-07-21:
Have you installed Thonny before?

Do you get this message right after starting Thonny or after you performed an action in Thonny?

Mitchell1944, 2020-07-26:
Sorry it has taken me awhile to get back to this!
I get the above as soon as I run the program.  I also tried installing in Windows 10 32bit on the same machine with the same result.
I don't know if this makes a difference but I am running an AMD processor.  Seemed to matter to Python-64AMD.
Also I should mention that I ran the recommended Windows install from the command line to 0 effect. In fact the above is after I ran that.
Really want this to work so thanks for the help. Best, Mitchell

aivarannamaa, 2020-07-26:
Can you please post Thonny's configuration file? I assume after you close the error message, you are able to operate Thonny's menus. Please select "Tools => Open Thonny data folder" and from the folder that gets opened, send me "configuration.ini"

Mitchell1944, 2020-07-26:
[general]
configuration_creation_timestamp = 2020-07-26T11:21:58.872998
language = en_US

[view]
full_screen = False
maximize_view = False
shellview.visible = True
assistantview.visible = False
astview.visible = False
stackview.visible = False
exceptionview.visible = False
filesview.visible = False
heapview.visible = False
helpview.visible = False
notesview.visible = False
objectinspector.visible = False
outlineview.visible = False
variablesview.visible = False
show_program_arguments = False
show_plotter = False

[layout]
zoomed = False
notebook_nw_visible_view = None
notebook_w_visible_view = None
notebook_sw_visible_view = None
notebook_s_visible_view = ShellView
notebook_ne_visible_view = None
notebook_e_visible_view = None
notebook_se_visible_view = None
width = 800
height = 650
left = 150
top = 15
west_pw_width = 199
east_pw_width = 199
nw_nb_height = 199
sw_nb_height = 199
s_nb_height = 199
se_nb_height = 199
ne_nb_height = 199

[file]
current_file = O:\Python Classes\Beginning\From 0 to 1\Strings-Python2.py
open_files = ['O:\\Python Classes\\Beginning\\From 0 to 1\\Strings-Python2.py']
recent_files = ['O:\\Python Classes\\Beginning\\From 0 to 1\\Strings-Python2.py']

[run]
program_arguments = 
dock_user_windows = False
pgzero_mode = False

Checked the DOS script again and corrected it for the actual location of the Python scripts and there is no 37-32, scripts are in 38
so no install script for Thoony.  Frankly I have installed so many Python apps I am not sure where any of it resides!
Hope this helps.
Best, Mitchell

Mitchell1944, 2020-07-27:
I am soooo sorry to have taken your time with this.  I found the problem was with my anti-malware program because the program was not telling me that it was blocking Thonny from running.  Cleared it and allowed the program to run and all seems to be OK!
Thanks so much for you patients and I am so happy that I will get to use it while I am learning Python. Again. thank you!
Best,
Mitchell

closed 2020-08-05 by Mitchell1944