thonny/thonny #1276: Version 3.2.7 debug icons not highlight and does not work and also the code complete when I type code

wrodusa2006, 2020-07-21:
Hi, I am Wald.

I have installed in my win 10 thonny Version 3.2.7 and the debug items do not highlight ar even from the menu drop list I can access them.

I did 2 trials uninstalling the program and installing again and the problem remain.

Also the code complete suggestions is not working.

ALL OTHER FEATURES WORK VERY WELL, I can program the esp32 without problems.

Please help to solve this issue, I have no idea about the problem.

Thanks...Wald

aivarannamaa, 2020-07-21:
Thank you for the report!

Thonny does not support debugging with MicroPython back-end. Code completion will be enhanced in next release

closed 2020-07-21 by aivarannamaa

wrodusa2006, 2020-07-21:
Hi...thanks for the feedback...regards...Wald