thonny/thonny #1277: Thonny crashing, related to jedi

ricardojosehlima, 2020-07-21:
Hi, 
   This is the message that I get when I try to open thonny (3.7.2):
Traceback (most recent call last):
  File "/home/ricardo/.local/lib/python3.6/site-packages/thonny/__init__.py", line 165, in launch
    bench = workbench.Workbench()
  File "/home/ricardo/.local/lib/python3.6/site-packages/thonny/workbench.py", line 167, in __init__
    self._load_plugins()
  File "/home/ricardo/.local/lib/python3.6/site-packages/thonny/workbench.py", line 373, in _load_plugins
    thonny.plugins.__path__, "thonny.plugins."  # type: ignore
  File "/home/ricardo/.local/lib/python3.6/site-packages/thonny/workbench.py", line 404, in _load_plugins_from_path
    getattr(m, load_function_name)()
  File "/home/ricardo/.local/lib/python3.6/site-packages/thonny/plugins/locals_marker.py", line 114, in load_plugin
    if jedi_utils.get_version_tuple() < (0, 9):
  File "/home/ricardo/.local/lib/python3.6/site-packages/thonny/jedi_utils.py", line 125, in get_version_tuple
    for item in jedi.__version__.split("."):
AttributeError: module 'jedi' has no attribute '__version__'

     The update of jedi from 0.17.0 to 0.17.2 caused this error. Solution: uninstall 0.17.2 and reinstall 0.17.0

aivarannamaa, 2020-07-23:
Thank you for the report!

Strange, to me it seems that Jedi 0.17.2 does have this attribute:

```
>>> import jedi
>>> jedi.__version__
'0.17.2'
```

aivarannamaa, 2020-07-23:
Can it be that you had a script named `jedi.py` in your working directory?

ricardojosehlima, 2020-07-23:
Yes, indeed there was, but not anymore. Thanks for the clarifications

aivarannamaa, 2020-07-24:
will be dealt with under #384

closed 2020-07-24 by aivarannamaa