thonny/thonny #1288: thonny's file explore cant find some folders.

croku, 2020-07-26:
same as the title...see the picure below
[image]

aivarannamaa, 2020-07-27:
It has different icon -- what kind of directory is this? Can you please post the output of following Python script when executed on your machine:

```
import os
print("listdir:", os.listdir("D:\\"))

path = "D:\\<insert the name of the problematic directory here>"
print("stat:", os.stat(path))

from ctypes import windll
print("atts:", windll.kernel32.GetFileAttributesW(path))
```

croku, 2020-07-28:
just a normal directory with a special icon？？？I dont know.
[image]

aivarannamaa, 2020-07-29:
GetFileAttributesW shows that this is marked as a system directory. Thonny currently hides system and hidden files and directories. In a future version I'll add an option to show also these files #1292.

If you don't want this directory to be treated as a system directory, then open system command prompt and enter

```
attrib "D:\<name of your dir>" -s
```

closed 2020-07-29 by aivarannamaa

croku, 2020-07-30:
that directory isnt a system directory.it is create by a software and I copied it from another disk.