thonny/thonny #1290: serial delay

aa2mz, 2020-07-26:
increasing delay in 
https://github.com/thonny/thonny/blob/master/thonny/plugins/micropython/serial_connection.py
line 54 (as you suggested) solved my device-file saving and reading problems.
I guessed and use ".03" now.

aivarannamaa, 2020-07-26:
Thanks! This is good to know!

aivarannamaa, 2020-10-28:
Obsoleted by a different approach in #1386

trfridle, 2020-10-28:
After troubleshooting a write issue on an ESP32 DevKit v4 module, I found that the serial rate was just too high through a CP20xx TTL to USB Serial controller.  Changing the value to .03 was the fix for this issue.

I will watch for a new build that uses the paste method to write and give it a test when it is available on this same hardware.