thonny/thonny #1291: https://github.com/thonny/thonny/releases/download/v3.2.7/thonny-3.2.7-armv7l.tar.gz fails

blaumeik51, 2020-07-27:
Hi Folks, 

I tried to update thonny to 3.2.7 on Linus, but it fails with a 404. 
https://github.com/thonny/thonny/releases/download/v3.2.7/thonny-3.2.7-armv7l.tar.gz --2020-07-27 15:21:21-- https://github.com/thonny/thonny/releases/download/v3.2.7/thonny-3.2.7-armv7l.tar.gz Auflösen des Hostnamens github.com (github.com)… 140.82.118.4 Verbindungsaufbau zu github.com (github.com)|140.82.118.4|:443 … verbunden. HTTP-Anforderung gesendet, auf Antwort wird gewartet … 404 Not Found 2020-07-27 15:21:21 FEHLER 404: Not Found

Am I doing something wrong?

aivarannamaa, 2020-07-27:
We don't provide binaries for arm architecture, only for 32-bit and 64-bit PC-s.

If you have version 3.2.6, then don't worry -- the changes in 3.2.7 only apply to macOS and Windows.

closed 2020-07-27 by aivarannamaa

blaumeik51, 2020-07-27:
I am running 3.2.7, but I have constantly backend crashes. So I read a bit and found out that updating to 3.2.8 should help. I tried it via the bash call, but it fails. Is there another way to get tge Version?

aivarannamaa, 2020-07-27:
There is no Thonny 3.2.8, the latest version is 3.2.7

> I have constantly backend crashes

Can you give me more details?

blaumeik51, 2020-07-27:
I am plotting an image in a loop with a time and date on it.so its updating every 5s using plt.pause(.1). After a few minutes it closes the figure and gives back an error stating tgat the backend crashed and I shall start, stop or reset. I can give you the exact statement later, but it is rather short, not very meaningfull.

blaumeik51, 2020-07-27:
The error is: Backend terminated or disconnected. Use 'Stop/Restart' to restart.
and the code on which it fails is:

import matplotlib.image as mpimg
import matplotlib.pyplot as plt
import datetime as dt
import locale
import time

locale.setlocale(locale.LC_ALL, '')

plt.style.use('dark_background')
plt.style.use('dark_background')
plt.rcParams['axes.linewidth'] = 0.1
plt.rcParams['toolbar']='None'
fig2 = plt.figure("Dashboard",figsize=(22.32,12.70))
fig2.set_tight_layout(False)
ax=fig2.add_subplot(111)
fig2.tight_layout()
start=True
while True:
    now=dt.datetime.now()

    if now.minute==6 or start:
        img=mpimg.imread('dashboard.png')
    ax.cla()   
    imgplot = ax.imshow(img)

    ax.text(500,40,now.strftime('%A') + ', ' + now.strftime('%d') + '. ' + now.strftime('%B') + '    ' + now.strftime('%H') + ':' + now.strftime('%M') + ' Uhr',fontsize=45)
    ax.axis('off')
    plt.pause(.001)
    start=False
    time.sleep(10)

It also happens if I run the code from the console.

Issue #910 seems to be somewhat similar.

aivarannamaa, 2020-07-27:
It looks like this program is leaking memory and I suspect the backend terminates because of running out of memory.