thonny/thonny #1296: slow when large dictionary open

bdkerman, 2020-08-02:
I've noticed when I have a very large dictionary open, running anything in Thonny is very slow and the processor is very busy.  Commands will eventually run, but the delay is a minute or so.  

this is not a memory constraint issue.  this is running on a machine with 64GB of RAM

The same code and dictionary does not cause a problem when running python without Thonny, or even in Jupyter.  

I'm imagining that it's something to do with the variable viewer or object inspector, but I can't seem to find a setting that alleviates the issue.  

thanks!

aivarannamaa, 2020-08-03:
Do you mean the directory is opened in Thonny's file browser (View => Files)?

How many items in this large directory?

Which operating system are you using?

bdkerman, 2020-08-04:
I mean python dictionaries.  A = {b:c}

When the dictionary is in memory, created or loaded from a pickle file.  
 The dictionary is  comprised of about 75k Pandas DataFrames.  Each, not that big.  

I’m running MacOS X Catalina.  Fully updated.

aivarannamaa, 2020-08-04:
Sorry, I misread your initial comment.

What do you mean by having "large dictionary open"? Printing it out? Opening in the Object Inspector? Or just being in the memory?

Does it perform better when you close Variables pane and/or Object Inspector (in case they are open)?

closed 2021-02-27 by aivarannamaa