thonny/thonny #1302: Autocomplete

abqmichael, 2020-08-10:
Sometimes when I push the tab key, Thornny errors out and raises an "autocomplete" error message. Clearing the message makes the dialog go away briefly then come back. As the error message can't be dismissed, the only way I've found to get out of this is to force quit Thornny and lose whatever changes I've made.

aivarannamaa, 2020-08-10:
Can you reproduce this? Which Thonny version are you using? On which platform?

abqmichael, 2020-08-10:
I’m not sure if I can replicate it. Though as it’s happened 3 times since I started using thonny (which was only a few weeks ago), I might get to a point where I can remember the steps that got me there.

I am using thonny on a 27” iMac running the latest macOS to develop on ESP32 boards which I plug in through the USB. The steps I remember was that I was moving back and forth in a Python program, moving some code from one method toward the bottom of the file (I was building some classes) to another toward the top. Of course when I cut code in one place, the syntax checker would go a bit wacky but not wanting to loose the copy string, I would leave that alone. Then went up to something simple, like adding an instance variable in the class definition and giving it an initial value. My normal “thing” is to put a comment on the line after the variable so it would look like this:

class foo:
	v1 = 10
	v2 = “quick brown fox” 	# just how quick will be determined
...

But when I pushed the tab, I think thonny looked at the mess way below and said “uh, oh, I can’t get that formatting right” and issued the warning. Well all I really know is it issued the warning. It’s also possible that I pushed the tab a couple times because the # wasn’t moving, but rather it seemed to be trying to reformat where v2 should line up.

I’ll try and watch closer.

Michael J.

abqmichael, 2020-08-10:
Oops

aivarannamaa, 2023-01-26:
Autocompletion got overhauled in Thonny 4. Closing as this exact problem has probably gone by now.

closed 2023-01-26 by aivarannamaa