thonny/thonny #1303: I cannot get the simple calculator to work

msuenson, 2020-08-12:
I am new to Python and started working the tutorials.
The example:

from calculator.simple import SimpleCalculator
my_calculator = SimpleCalculator()  
my_calculator.run('2 * 2')
print(my_calculator.lcd) 

gives an error:

 File "C:\Users\... \Python\calculator.py", line 1, in <module>
    from calculator.simple import SimpleCalculator
  File "C:\Users\...\Python\calculator.py", line 1, in <module>
    from calculator.simple import SimpleCalculator
ModuleNotFoundError: No module named 'calculator.simple'; 'calculator' is not a package

When I look at the designated repository, I find the folder "calculator"
And in the Calculator folder, there is a "simple.py"

And in simple.py, I find:

# -*- coding: utf-8 -*-

"""Simple calculator.
"""
# Simple calculator library
import math

class SimpleCalculator():
    """Simple calculator class."""
.....

aivarannamaa, 2020-08-13:
You seem to have both directory `calculator` and module `calculator.py`. This confuses Python. You should rename your main script.

aivarannamaa, 2020-08-13:
Somewhat related to #384

msuenson, 2020-08-13:
No, the folder is named "calculator". The module in the folder "calculator" is called simple.py.

But when I lifted the module up one level to the directory containing the folder "calculator" it worked.

closed 2020-08-13 by aivarannamaa