thonny/thonny #1304: Dracula, a drak theme for Thonny

danspinola, 2020-08-13:
Hi! 👋

I just wanted to say that I have ported the popular dark theme Dracula 🧛‍♂️ for Thonny and it was accepted into the Dracula organization.

It's already indexed to PyPI as `thonny-dracula`.

Honestly, I find myself using Thonny more often than PyCharm, especially for quick scripts. Thanks for the great software. 🙌

https://draculatheme.com/
https://github.com/dracula/thonny

aivarannamaa, 2020-08-14:
Thank you! Looks good! :)

It's good to see people adding Thonny plug-ins.

BTW, `__init__.py` should be omitted from `thonnycontrib`, because it is meant to be a [namespace package](https://packaging.python.org/guides/packaging-namespace-packages/#native-namespace-packages).

danspinola, 2020-08-14:
Thanks! I'll remove it.

closed 2020-08-14 by aivarannamaa