thonny/thonny #1306: focus after run script

denise-amiga, 2020-08-19:
you could return the focus to the shell, after running a script?

thx

aivarannamaa, 2020-08-20:
I believe many users prefer current approach -- they can quickly continue editing the script after running.

Why do you prefer moving the focus to the shell?

denise-amiga, 2020-08-20:
sorry, I'm new to python, the point is that I usually make several versions of the algorithm in the editor, and then test them in the shell with different arguments/parameters.
The problem is that when I click with the mouse on the "play" button the focus is lost, whether you are in the editor or in the shell.
if you press "F5" everything works perfect, keep the focus

aivarannamaa, 2020-08-20:
I agree that there is no point in setting the focus to the toolbar button, but that's how Tkinter works.

I recently changed it so that editor is given focus after a toolbar button gets clicked (this will appear in the next release). I'll try to make it smarter so that the previous focus gets restored.

denise-amiga, 2020-08-20:
OK thank you very much

aivarannamaa, 2020-08-21:
Let's keep it open for now as I may be able to enhance it more.

aivarannamaa, 2020-12-12:
Closing for now, as there are too many more pressing issues in the queue.

closed 2020-12-12 by aivarannamaa