thonny/thonny #1311: Thonny 64-bit doesn't work with matplotlib

bryanosborne, 2020-08-26:
Running Win10.  Installed matplotlib 3.3.1 and numpy 1.19.1 via Tools>Manage Packages.  Attempted to run:

import matplotlib.pyplot as plt
import numpy as np

x = np.linspace(-1, 1, 50)
print(x)
y = 2*x + 1

plt.plot(x, y)
plt.show()

Received:
>>> %Run testplot.py
Traceback (most recent call last):
  File "D:\MyFiles\ORU\CSC111-IntroToComputing\Python\testplot.py", line 1, in <module>
    import matplotlib.pyplot as plt
  File "C:\Users\josborne\AppData\Roaming\Python\Python37\site-packages\matplotlib\__init__.py", line 107, in <module>
    from . import cbook, rcsetup
  File "C:\Users\josborne\AppData\Roaming\Python\Python37\site-packages\matplotlib\cbook\__init__.py", line 28, in <module>
    import numpy as np
  File "C:\Users\josborne\AppData\Roaming\Python\Python37\site-packages\numpy\__init__.py", line 138, in <module>
    from . import _distributor_init
  File "C:\Users\josborne\AppData\Roaming\Python\Python37\site-packages\numpy\_distributor_init.py", line 26, in <module>
    WinDLL(os.path.abspath(filename))
  File "C:\Users\josborne\anaconda3\lib\ctypes\__init__.py", line 364, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is not a valid Win32 application

Seems to work ok on MacOS

aivarannamaa, 2020-08-26:
One possible explanation: you installed numpy via Thonny while Thonny was configured to use its bundled 32-bit Python, ie 32-bit numpy was installed to user site-packages. Later you switched Thonny to use 64-bit Anaconda, but the 32-bit numpy in user site-packages takes precedence and hence the conflict.

Could it be like this?

aivarannamaa, 2020-08-26:
If you intend to use Anaconda as your back-end, then it makes sense to uninstall numpy and matplotlib from Thonny. This will unshadow Anaconda's numpy and matplotlib.

bryanosborne, 2020-08-27:
Spot on @aivarannamaa and thanks.  Switched machines and forgot that Anaconda was set as the interpreter.  Silly mistake!

Thanks again!

closed 2020-08-27 by bryanosborne