thonny/thonny #1312: Thonny V3.2.7 sees my circuitpython device as a MicroPython device

bodat86, 2020-08-26:
Whenever I attempt to load a file from my Circuitpython device, Thonny asks me whether I want to load from This computer or MicroPython device. If I choose 'MicroPython device', I get an error: 'Device is not connected'. If I choose 'This computer', I can open my "J Drive" (the one my Circuitpython device is connected to), and I'm able load the file. Through all this my REPL (from the Circuitpython device) is running fine when I first start Thonny. 

I am using Win10 and Adafruit Metro M4

Under Tools/options/Interpreter is set to  CircuitPython(generic) and the Port is set to the correct "USB Serial Device (COM8)". 

I've reinstalled Thonny, changed the Interpreter to "The same interpreter which runs Thonny (default)" and back to CircuitPython (generic). All to no avail.

Any suggestions???

aivarannamaa, 2020-08-27:
Thonny is supposed to derive "MicroPython" vs "CircuitPython" label from the text that appears when you enter the REPL (eg. "Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit Trinket M0 with samd21e18"). Does your device show the word "CircuitPython" there?

Do you only get "Device is not connected"? No stacktrace or anything else? Can you see CircuitPython files in Thonny's file browser (View => Files)?

bodat86, 2020-08-27:
Hi! and thank you for your response!

I restarted my computer first thing this morning and Viola! It's working now???? I had restarted it yesterday and it did not work???

Thanks for your interest in my issue!

Bodat

aivarannamaa, 2020-08-27:
MicroPython (and CircuitPython) support in current Thonny version is not very robust. I hope to release beta of next version soon, with refactored and more robust support.

closed 2020-08-27 by aivarannamaa