thonny/thonny #1313: Acess to REPL not possible if SLEEP_LIGHT is selected

wnelis, 2020-08-30:
Using Thonny 3.2.7, a build of micropython of 2020.08.29, and a D1 mini ESP8266 development board, a program is written to read the voltage on the analogue port and report the value read to an MQTT broker. This program is waiting most of the time: every 10 seconds a measurement is done and every 300 seconds a summary of 30 measurements is published. Since the program is extended with SLEEP_LIGHT, using `esp.sleep_type( esp.SLEEP_LIGHT )`, it is no longer possible to enter REPL. The only way I've found to test a new version of the program is to erase flash on the ESP8266, (re)install micropython and (re)install all Python modules.

The use of SLEEP_LIGHT causes the current flow to decrease while sleeping from about 25 [mA] to about 10 [mA]. As the ESP8266 will be powered from a source with little spare I like to minimize the power consumption, however without resorting to a deep sleep. (State information is lost at deep sleep, but in this case the statistics of the measurements need to be saved across a sleep period.)

After a power cycle of the ESP8266, Thonny does read some data via the USB port:

```
ip:192.168.178.29,mask:255.255.255.0,gw:192.168.178.1
Wifi connection established.
MQTT broker connection established.
pm open,type:1 0
```

After clicking on 'Stop/Restart backend':

```
─────────────
Could not enter REPL. Trying again with 1 second waiting time...
Could not enter REPL. Trying again with 3 second waiting time...
Could not enter REPL. Trying again with 5 second waiting time...
Could not enter REPL. Giving up. Read bytes:
b'Traceback (most recent call last):\r\n  File "main.py", line 228, in <module>\r\n  File "main.py", line 131, in Publish\r\n  File "millisDelay.py", line 151, in justFinished\r\n  File "millisDelay.py", line 119, in _md_try_to_sleep\r\nKeyboardInterrupt: \r\n\r\nMicroPython v1.12-691-g0c7354afa on 2020-08-29; ESP module with ESP8266\r\nType "help()" for more information.\r\n>>> ^NH^H\x1b[5D'

Your options:

  - check connection properties;
  - make sure the device has suitable firmware;
  - make sure the device is not in bootloader mode;
  - reset the device and try again;
  - try other serial clients (Putty, TeraTerm, screen, ...);
  - ask for help in Thonny's forum or issue tracker.

Backend terminated or disconnected. Use 'Stop/Restart' to restart.
```

In other attempts to connect to REPL, Thonny reported a crash due to a time-out.

Some documentation states that an ESP8266 is less responsive when in SLEEP_LIGHT mode. However, I have not found any documentation stating that it's use will prohibit the IDE to enter REPL. So I am wondering, is this an issue of Thonny or is this behavior inherent to the use of SLEEP_LIGHT?

aivarannamaa, 2020-08-31:
Have you tried entering the REPL with another tool (eg. Putty if you are using Windows or tio in Linux)?

From the traceback it looks like Thonny read some garbage while looking for the prompt: `>>> ^NH^H\x1b[5D`. This garbage is the reason Thonny didn't detect the prompt. Any ideas what was generating it?

wnelis, 2020-09-01:
The answer to your last question is no. Moreover, it is not reproducible. The program contains only a few print statements, each containing simple ASCII text. The only other type of output are messages to an MQTT broker, but those should not appear on the serial port.

The answer to your first question is also no, so I've tried it. I'm using Linux Mint 20 and minicom v2.7.1. The program on the ESP8266 has been extended to output a single 'M' (measure) each 10 seconds, and a 'P' with end-of-line (publish) every 100 seconds. When using minicom, the periodic messages, 'M' and 'P', do appear. I assume the only way to enter REPL is by typing ^C. The output of minicom in this test is as follows:

```
Welcome to minicom 2.7.1

OPTIONS: I18n 
Compiled on Dec 23 2019, 02:06:26.
Port /dev/ttyUSB0, 21:10:21

Press CTRL-A Z for help on special keys                 
                                                        
MMMMMMMP
MMMMMMMMMMP
MMMMMTraceback (most recent call last):
  File "main.py", line 229, in <module>
  File "main.py", line 119, in Measure
  File "millisDelay.py", line 151, in justFinished
  File "millisDelay.py", line 119, in _md_try_to_sleep
KeyboardInterrupt: 
��ɽA�ѡ塽��v1.12 on 2019-12-20; ESP module with ESP8266
Type "help()" for more information.
>>> 
>>>
```

I've typed ^C multiple times before the trace-back was shown. When the `>>>` was shown, enter was typed multiple times before another prompt was shown.

Then the ESP8266 was powered off and on. The print output of the program as shown by minicom is:

```
Wifi connection established.
MQTT broker connection established.
MMMMMMMMMP
MMMMMMMMMMP
MMMMMMMMMMP
```
After typing ^C 13 times (!), the trace-back appeared as well as the prompt.

```
Traceback (most recent call last):
  File "main.py", line 229, in <module>
  File "main.py", line 119, in Measure
  File "millisDelay.py", line 151, in justFinished
  File "millisDelay.py", line 119, in _md_try_to_sleep
KeyboardInterrupt:

MicroPython v1.12 on 2019-12-20; ESP module with ESP8266
Type "help()" for more information.
>>>
```

After entering a new-line (enter) for 3 times, another prompt appeared. The REPL is thus hardly responding. When I typed `print('Hello')` only two letters were echoed to the screen.

After stopping mincom, a power-cycle of the ESP8266, starting Thonny and hitting the `stop` button, there was an immediate response:

```
MicroPython v1.12 on 2019-12-20; ESP module with ESP8266
Type "help()" for more information.
>>> print('A')
PROBLEM WITH THONNY'S BACK-END:

Traceback (most recent call last):
  File "/home/wim/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 340, in handle_command
    response = handler(cmd)
  File "/home/wim/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 697, in _cmd_execute_source
    _, _, value_repr = self._execute_print_expr(cmd.source)
  File "/home/wim/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 466, in _execute_print_expr
    return self._execute(script, capture_output)
[... 21 lines cut ...]
  File "/home/wim/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/connection.py", line 35, in read
    self._read_buffer.extend(self._read_queue.get(True, timer.time_left))
  File "/home/wim/apps/thonny/lib/python3.7/queue.py", line 178, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/wim/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 340, in handle_command
    response = handler(cmd)
  File "/home/wim/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 713, in _cmd_get_globals
    "{name : repr(value) for (name, value) in globals().items() if not name.startswith('__')}"
  File "/home/wim/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 469, in _evaluate
    _, _, value_repr = self._execute_print_expr(expr, prelude, cleanup)
  File "/home/wim/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 466, in _execute_print_expr
    return self._execute(script, capture_output)
  File "/home/wim/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 438, in _execute
    ok = self._connection.read(2)
  File "/home/wim/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/connection.py", line 37, in read
    raise TimeoutError("Reaction timeout. Bytes read: %s" % self._read_buffer)
TimeoutError: Reaction timeout. Bytes read: bytearray(b'')
```

If I repeat this test, some variations on the error messages above appear, all starting with `PROBLEM WITH THONNY'S BACK-END`.

aivarannamaa, 2020-09-02:
I hope to release first beta of Thonny 3.3 tonight with refactored MicroPython support. Maybe this will help you.

BTW, is your code multithreaded?

wnelis, 2020-09-02:
No, the program is not multi-threaded.

wnelis, 2020-09-03:
The tests above show that (A) using SLEEP_LIGHT makes the ESP8266 unresponsive and (B) that the back-end of Thonny may crash on an unresponsive system. For the purpose of development and tests, it seems necessary to avoid SLEEP_LIGHT.

Is it possible to detect that the IDE is connected to ESP8266? If so, the invocation of `esp.sleeptype` can be set to either SLEEP_LIGHT in case of an operational run (no IDE attached) or SLEEP_MODEM in case of development tests (an IDE attached).

wnelis, 2020-09-03:
A local version of Thonny version 3.3.0-b2 is installed using `pip3 install --pre --user --upgrade thonny`. (Thonny version 3.2.7 is installed using `sudo apt-get install thonny`.) It is still not possible to enter REPL if the ESP8266 is using SLEEP_LIGHT mode.

Most often nothing is read and Thonny reports it is giving up. Sometimes some text is read, but still Thonny is giving up:
```
Could not enter REPL. Giving up. Read bytes:
b'Traceback (most recent call last):\r\n  File "main.py", line 229, in <module>\r\n  File "main.py", line 119, in Measure\r\n  File "millisDelay.py", line 151, in justFinished\r\n  File "millisDelay.py", line 119, in _md_try_to_sleep\r\nKeyboardInterrupt: \r\n\r\nMicroPython v1.12 on 2019-12-\x02\x04\xda\x02*MA\x81module with ESP8266\r\nType "help()" for more information.\r\n>>> \r\nraw REPL; CTRL-B to\x00\x14\x16'
```

aivarannamaa, 2020-09-04:
I can't think of any clean way the program could detect that it is run from an IDE.

Could you send me a minima program for reproducing this situation? I would try whether I can wake the device with serial BREAK. If not, I would ask in MicroPython forum.

aivarannamaa, 2020-09-04:
I found an idea which may solve your case: https://forum.micropython.org/viewtopic.php?f=16&t=3623

In short -- leave some time for interruption before you start your work involving SLEEP_LIGHT

wnelis, 2020-09-04:
Below a minimal program showing the issue is included. The idea in the above mentioned forum topic is incorporated. Tests have shown that this is indeed a good escape: if REPL cannot be entered from Thonny, a hardware reset combined with a timely hit of the `stop` button will cause REPL to be entered.

```
import esp
import machine
import network
import time

#
# Global variables and initialization.
# ------------------------------------
#
ledlvl= 0

# Give a time window after a reset to enter REPL.
print( 'Initial delay of 5 [s] before program starts.' )
time.sleep( 5 )
print( 'Starting program.' )

# Switch to a lower power mode when sleeping.
esp.sleep_type( esp.SLEEP_LIGHT )

# Do something visible, blink the onboard LED.
led= machine.Pin( 2, machine.Pin.OUT )

# Switch off the internal access-point.
station= network.WLAN( network.AP_IF )
station.active( False )

#
# Main program.
# -------------
#
while True:
    ledlvl= 1 - ledlvl
    led.value( ledlvl )
#   machine.lightsleep( 1000 )
    time.sleep_ms( 1000 )
```

Note that if access point mode is not deactivated, the issue is not reproducible. Thus the combination of SLEEP_LIGHT mode and the lack of an access point function seem to trigger the issue.

The stated problem is solved: a time window can be created on demand in which Thonny can enter REPL, while power usage is minimized.

aivarannamaa, 2020-09-20:
I'll close the issue, because I don't how to improve Thonny to support this case.

A related thing: In latest beta (3.3.0b3) Thonny adds soft-reboot command (Ctrl+D) among the actions taken by "Stop/Restart" buttons. (Just mentioning, I suppose it won't help in your case.)

closed 2020-09-20 by aivarannamaa