thonny/thonny #1326: Where are all the commands?

mostmarc, 2020-09-06:
installing esptool gives no install option. 

Thonny 3.2.7
win 10 64bit
Python 3.8.3 64bit
Tk 8.6.9

esp32
C:\Users\marcus\esptool\__pycache__>

aivarannamaa, 2020-09-06:
Sorry, I don't understand what are you trying to do.

mostmarc, 2020-09-06:
Thanks for the swift reply,

I have installed Thonny IDE, working but he cannot find all the tools and
or modules for ESP32. Following the instructions on Youtube video and the
PDF resources, I can’t find any suggestions for search path installation.
The device tab has no access to the menu. The debugger is also in
non-active status. Even reinstalling and upgrading to Python 3.8 and eps
amends gives no solution. The “manage plug in menu” gives no button to
activate the choice made.

Thanks in advance

Marcus

mostmarc, 2020-09-06:
[image: image.png]
Select device tab

mostmarc, 2020-09-06:
[image: image.png]

aivarannamaa, 2020-09-06:
> I can’t find any suggestions for search path installation.

Sorry, I don't understand this one

> The device tab has no access to the menu. 

Do you mean there is no "Device" menu (or nothing in the "Device" menu)? Latest Thonny versions don't use it.

> The debugger is also in non-active status.

Debugger is not available for MicroPython back-end

I don't see the pictures you posted (it seems that GitHub simply drops pictures sent by e-mail)

mostmarc, 2020-09-06:
as attachment

aivarannamaa, 2020-09-06:
GitHub doesn't forward attachments. You need to use GitHub web interface if you want to post pictures: https://docs.github.com/en/github/managing-your-work-on-github/file-attachments-on-issues-and-pull-requests

mostmarc, 2020-09-07:
[image]

aivarannamaa, 2020-09-07:
Did you read the message you got from the Device menu? It says what is going on. I don't know how to explain it better. Maybe this page helps: https://github.com/thonny/thonny/wiki/MicroPython

The error "Could not ensure raw prompt" may be fixed in the beta of Thonny 3.3. I recommend you try this: https://github.com/thonny/thonny/releases/tag/v3.3.0b2

I still don't understand what do you want to do with esptool and what is the problem

mostmarc, 2020-09-07:
https://www.youtube.com/watch?v=lvmNLuHj25o
~minute 3.30 esptool!!

aivarannamaa, 2020-09-07:
I see -- you want to install esptool for Thonny. But what's the problem? From your screenshot I see that you have esptool in a wrong location, Thonny won't find it from there. Did you use "Tools => Manage **plug-ins**" ?

mostmarc, 2020-09-07:
Yes that is the question, where of what are the path stelling parameters in
Thonny and how to change them? So you can synchronize the workpage and or
libs of modules.

aivarannamaa, 2020-09-07:
Sorry, I didn't understand your last message either. What is your native language? Try asking in your native language, maybe I'll understand.

mostmarc, 2020-09-07:
How to change the search path in Thonny? For the workspace and the module
folder.

aivarannamaa, 2020-09-07:
If you mean PYTHONPATH, then Thonny does not provide any means for setting or changing it.

If you want to use another interpreter instead of the bundled one, then see "Run => Select interpreter"

If you are having trouble with esptool, then you could try the beta of Thonny 3.3, which includes esptool: https://github.com/thonny/thonny/releases/tag/v3.3.0b2

mostmarc, 2020-09-07:
Looks interesting, going to try it.

Thanks in advance.

closed 2020-09-20 by aivarannamaa