thonny/thonny #1327: tweaking the "Manage packages" dialog

aoloe, 2020-09-08:
In my eyes it's hard to follow what happens in the "Manage packages" dialog.

It's very simple to install a package with it and that's perfect!  
But for a new users it might be hard to understand where to click and what has happened after the click.

Here are a few ideas:

- When clicking on the links in the `<INSTALL>` text I get an error and I can't test what they do.  
  But I'm not sure that they are helpful for the target users of Thonny. I would move the features in a separate dialog (that can be reached by a button in this dialog or from another menu item).
- The list of packages in the left should extend to the top of the dialog. The text field for the find feature, should start on the right of it.
- The "Find" button could just say "Find" (if the widgets support it, you can add a tooltip with the details, but I'm not sure most users read them).
- Remove the "<INSTALL>" entry in the list. If a package is found, just deselect all items in the list.
- When installing, the popup with the log text can be very slow or not scroll at all (_freezing_ for many seconds): it's not clear that I should keep it open during the install.  
  During the installing, I would:
  - change the label of "Install" to "Installing..." and deactivate the button;
  - change the label of "Close" to "Cancel" and get the button to cancel the install process (if it's not possible, add a "Cancel" button next to "Install" and hide "..." during the install process)
  - do not show the install log in a popup but in a text field in the dialog itself (below the package description or instead of it).

aivarannamaa, 2020-09-20:
Excuse me for late reply!

> When clicking on the links in the <INSTALL> text I get an error and I can't test what they do.

Indeed, a bug has slipped into handler of the first link (fixed now under #1344). Did you get errors from other links as well?

> But I'm not sure that they are helpful for the target users of Thonny. I would move the features in a separate dialog (that can be reached by a button in this dialog or from another menu item).

I'm not sure about "Install from local file", but "Install from requirements.txt" may be relevant in school context, where teacher has prepared a requirements.txt with required packages (possibly with specific versions) to be installed by students on their machines. And as these two features are close cousins, I'd leave them where they are.

> The list of packages in the left should extend to the top of the dialog. The text field for the find feature, should start on the right of it.

Why? 

> The "Find" button could just say "Find" (if the widgets support it, you can add a tooltip with the details, but I'm not sure most users read them).

What's wrong with "Search on PyPI"? If it is grammatically wrong, then I'm happy to fix it (I'm not good with English prepositions).

> Remove the "" entry in the list. If a package is found, just deselect all items in the list.

What do you mean by `"" entry` ?

> it's not clear that I should keep it open during the install.

Isn't the button "Cancel" enough to signify the meaning of the install dialog?

aoloe, 2020-09-22:
1. yes i also get an error from the "install from local file". i have not tried the newest build, though.
2. i understand that using a requirements.txt is a very useful feature. my issue is with the way it's presented: the link is not easy to find in a class context and "noise" for the others... see below for a UI proposal.
3. at first i wanted to say: personal taste. then i looked at my screen and saw the mail and chat program doing exactly that: a list on the left extending to the top and several fields on the right.
    of course it's a detail, but at first i did not really get where i had to click and what happened when i pressed the find button.
4. where should it search if not on PyPI? Thonny offers only one way to search for packages, so the detail is not needed in the button (in my eyes the text in the buttons should be kept as short as possible... with exceptions, of course!)
    (while creating the mockup below i also noticed that i would prefer the button to say "Search", not "Find"... again: probably it's because I'm used to that...)
5. the "" entry has been eaten by markdown, sorry it's the INSTALL with a less than and greater than
6. no, the "Cancel" button is too similar to a "Close" button. since there can be no output for a long time in the dialog, and it's IMO not 100% clear that the dialog is giving the install status, i just wanted to close the pop up because i thought it was some "stupid" message.
    of course, i'm known for being against pop ups : - )

voilà, all this having been said, here is a quick mockup that shows my thoughts.

[image]

feel free to use it in full, in parts or just ignore it.

personally, i don't use thonny, but i gave it to somebody i'm mentoring, and i will keep an eye how well she gets around with the UI : - )

aoloe, 2020-09-22:
p.s. i just noticed an error: the two buttons in bottom left corner should say "Choose", not "Install"... but it's not so important.

aivarannamaa, 2020-09-22:
Thank you for this effort!

The shorter search box indeed looks good in your picture. I'll consider this and your other suggestion when I'm done with Thonny 3.3 and start preparing version 4.0 (which probably also brings other changes to package management)

aivarannamaa, 2024-05-30:
The dialog has been moved to a view and reworked (#3151). You can see the result in current master state or in the upcoming 5.0.

closed 2024-05-30 by aivarannamaa