thonny/thonny #1328: window size adjustment moves wrong direction

zxsq-cc, 2020-09-08:
OS: macOS mojave
Thonny version: 3.1.2
Dragging window edge from the left side moves the right side of the window in/out instead of the left side. Other sides work as expected.
(screen recording: https://vimeo.com/455811217)

aivarannamaa, 2020-09-08:
Thank you for the report!

Could you please try [latest version (3.2.7)](https://github.com/thonny/thonny/releases/download/v3.2.7/thonny-3.2.7.pkg) or the [beta of 3.3.0](https://github.com/thonny/thonny/releases/download/v3.3.0b2/thonny-3.3.0b2.pkg)?

zxsq-cc, 2020-09-08:
Still an issue in 3.2.7, and in 3.3.0 beta.

aivarannamaa, 2020-09-09:
I can't reproduce this in Catalina and I don't have any ideas what may cause it. As it doesn't look like a serious issue, I'll leave it for now

MotusPerpetuusSprocketus, 2020-10-02:
@aivarannamaa 
I hope this helps display the issue more throughly. I don't know if the requested improvements are even possible. Thanks for making an (otherwise) awesome program.

[Thonny vs Mac.m4v.zip](https://github.com/thonny/thonny/files/5320260/Thonny.vs.Mac.m4v.zip)

aivarannamaa, 2020-10-03:
@MotusPerpetuusSprocketus , thank you for this video!

Regarding tiling and full screen -- this should be fixed in newest Tk, but unfortunately newest Pythons don't use this. I'll see what can be done.

aivarannamaa, 2021-02-17:
Closing this for now, as I'm not able to reproduce. Hopefully will be fixed in Thonny 4.0

closed 2021-02-17 by aivarannamaa