thonny/thonny #1330: Packages cant install because of SSL

nheyland, 2020-09-11:
Is there any way I can get around this? Been searching for hours cant seem to find anything that works? OS Big Sur if that helps at all? 

[image]

aivarannamaa, 2020-09-12:
"SSL certificate is not validated" is just a warning, the real problem lies elsewhere -- this package (`requests`) is meant for CPython, not MicroPython.

aivarannamaa, 2020-09-12:
BTW, are you using Big Sur? My Mac is not compatible, so I haven't tested Thonny with it. Is everything else running smoothly?

nheyland, 2020-09-12:
Ahh that's makes sense, new here just getting the hang of everything in micropython. Yes this is Big Sur. I had trouble opening the package manager in the previous version but now I am in the latest Thonny build, and so far no issues, let me know if you need me to try anything specific out for you for testing.

closed 2021-01-19 by aivarannamaa

tony1tf, 2021-01-24:
I’m having exactly the same problem with Thonny on Big Sur trying to add numpy to my RPi Pico. Did you find a solution?

aivarannamaa, 2021-01-24:
@tony1tf, numpy doesn't work on MicroPython devices.

tony1tf, 2021-01-24:
Wow! - thanks for that - that's a lot of head scratching finished.
Tony