thonny/thonny #1333: Unconnected network drive shorcuts make Files explorer broken

codemee, 2020-09-12:
If theres's network drive shorcuts that cannot be connected on Windows, click the **This Computer** button in the **Files** explorer would raise the FileNotFoundError exception so that I cannot view files anymore.  

[image]

aivarannamaa, 2020-09-12:
Thank you for the report! I'll fix it in the next beta of 3.3

You should be get out of this state when you enter following command into shell:

`%cd C:\`

aivarannamaa, 2020-09-12:
Sorry, my previous suggestion doesn't work with MicroPython backend. Switch to Python 3, issue this command and then switch back to MicroPython.

codemee, 2020-09-14:
Thanks, but the command should be %cd C:\\ to escape from '\'.

closed 2020-10-28 by aivarannamaa