thonny/thonny #1335: breakpoints / run to cursor fail

Flion, 2020-09-13:
Using Thonny 3.2.7 (latest on download page) and breakpoints are sometimes totally ignored using debug current script (either faster or nicer).  I was trying to debug a loop, put the breakpoint at the top of the loop or inside the loop and the program runs to end. Sometimes the breakpoint hits in first iteration, ignored thereafter. Similarly, if I select run to cursor, it runs to the cursor and stops.  If I select it again (for the next iteration of the loop) it runs to the end of the script. Attached is an example script and associated library.  Yes, I know, not pythonic; I am teaching array (not list) basics to beginning programmers. I'd still expect the debugger to work with this code.
[SelSort.zip](https://github.com/thonny/thonny/files/5214756/SelSort.zip)

aivarannamaa, 2020-09-14:
Thank you for the report!

Could you please post a simpler script, which demonstrates this issue, together with breakpoint locations?

aivarannamaa, 2020-09-14:
Actually, the same script is also OK, if you tell me exactly the positions of breakpoints/cursor and the steps to reproduce the problem.

Flion, 2020-09-14:
My apologies, I discovered the problem with breakpoints; I was selecting the 'run script' icon instead of the 'resume' icon.  The run to cursor action, though, is still not working.  For example, I can start the debugging session and select line 9 in the selectionSortTest.py script by right clicking.  Choosing run to cursor has the debugger advance to that line.  Repeating the action causes the script to run to the end.  Thinking that trying to stop at the top of the while loop (line 9) might be a problem, I used line 10 instead, with the same result.  Lines 14 (top of for loop) and 16 (inside for loop) exhibit the same behavior.  I also thought to try to single step into the selection sort method, to open the function call window and try to invoke run to cursor there (same locations) but then the run to cursor action is completely ignored.

To reiterate, the breakpoint problem was user error but run to cursor seems to have a problem.

aivarannamaa, 2020-09-20:
I could reproduce the problem when running to cursor at line 9, but on line 10 I could run to cursor repeatedly and each time one interation of the while loop was performed.

Which platform are you using?

Flion, 2020-09-20:
Windows 10 64-bit on a Microsoft Surface Pro. About Thonny reports version 3.2.7 and Python 3.7.7 (32-bit) and Tk 8.6.9

aivarannamaa, 2020-11-07:
I made some changes to the debugger, and could not reproduce this problem anymore. Version 3.3 will be released in a week. Please reopen if the issue is still not fixed for you.

closed 2020-11-07 by aivarannamaa