thonny/thonny #1336: An inexplicable error report

JKP-feiyu, 2020-09-14:
The copied code is inexplicably wrong. In fact, there is no syntax error, but it shows a syntax error

JKP-feiyu, 2020-09-14:
This is the code and the error
`'''
说明：编程实现MQTT通信，连接到阿里云，周期性上报温度数据。
'''
import network,time
from simple import MQTTClient #导入MQTT板块
from machine import Timer

#WIFI连接函数
def WIFI_Connect():
    wlan = network.WLAN(network.STA_IF) #STA模式
    wlan.active(True)                   #激活接口
    start_time=time.time()              #记录时间做超时判断

    if not wlan.isconnected():
        print('connecting to network...')
        wlan.connect('JKP', '12345678') #输入WIFI账号密码

        while not wlan.isconnected():
            #超时判断,15秒没连接成功判定为超时
            if time.time()-start_time > 10 :
                print('WIFI Connected Timeout!')
                break

    if wlan.isconnected():
        print('network information:', wlan.ifconfig())
        return True
    else:
        return False

#发布数据任务
def MQTT_Send(tim):
    send_mseg = '{"params":{"PowerSwitch_1": 0},"method": "thing.event.property.post"}'
    client.publish(topic="/sys/a1ve2pDFlWK/Defend/thing/event/property/post",
                    msg=send_mseg,qos=0, retain=False)

def sub_cb(topic, msg):
    print((topic, msg))

#执行WIFI连接函数并判断是否已经连接成功
print(WIFI_Connect())
if WIFI_Connect():
    #设备三元组信息
    ProductKey= '***'
    DeviceName = "Defend"
    DeviceSecret = "88f579bd110edbee65ee40bb2394e37a"#使用你自己的

    #阿里云设备令牌
    #客户端ID：123456,建议是MAC或唯一
    CLIENT_ID = "Defend|securemode=3,signmethod=hmacsha1,timestamp=789|"
    #用户名：DeviceName&ProductKey
    user_name = "Defend&a1ve2pDFlWK"
    #用户密码：DeviceSecret,clientId123456deviceNameDHT11productKeya1mFuprkAAAtimestamp789
    user_password = 'D0ECF0BE0A2849AB695A69224DB4320C21D745F1'#使用你自己的

    #阿里云IP和端口
    SERVER = "a1ve2pDFlWK.iot-as-mqtt.cn-shanghai.aliyuncs.com"
    PORT = 1883

    #TOPIC = '/public/01Studio/1' # TOPIC名称
    client = MQTTClient(client_id=CLIENT_ID, server=SERVER, port=PORT, user=user_name, password=user_password, keepalive=60)
    client.set_callback(sub_cb)                         #set callback
    client.connect()

    #开启RTOS定时器，编号为-1,周期20000ms，执行MQTT发布任务
    tim = Timer(-1)
    tim.init(period=20000, mode=Timer.PERIODIC,callback=MQTT_Send)

`
Traceback (most recent call last):
  File "<stdin>", line 15
SyntaxError: invalid syntax

aivarannamaa, 2020-09-14:
It looks like some bits were not transferred properly to the device.

Which device and which Thonny version are you using?

Can you execute it by saving to main.py and soft-rebooting your device (Ctrl+D) ?

JKP-feiyu, 2020-09-16:
He doesn't execute it, he just goes to the micropython command line

aivarannamaa, 2020-09-16:
Which device and which Thonny version are you using?

JKP-feiyu, 2020-09-16:
ESP32  
esp32-idf3-20191220-v1.12.bin

aivarannamaa, 2020-09-16:
And Thonny version?

JKP-feiyu, 2020-09-16:
3.2.7

JKP-feiyu, 2020-09-17:
Excuse me, is there any good solution to this problem.

aivarannamaa, 2020-09-17:
Please try latest beta: https://github.com/thonny/thonny/releases/tag/v3.3.0b2

There I have used smaller block size by default and this may cure this problem.

JKP-feiyu, 2020-09-17:
I don't understand what "used small block size by default" means.

aivarannamaa, 2020-09-17:
> I don't understand what "used small block size by default" means.

It means Thonny sends data to the device in smaller chunks and waits a bit between the chunks in order to reduce the likelihood of overflowing device's input buffer.

closed 2021-01-19 by aivarannamaa