thonny/thonny #1338: Thonny / squeezed text - extremely slow & crashes

tiomanisland, 2020-09-14:
I came across Thonny because I want to learn Python. What a nice little tool this is! It works nice and smooth but there is one problem. I wrote a little script that pulls HTML from source and when I display the squeezed result (HTML) and click on "Details" a small popup will open with text. This is ok but even if text is small (32kb) everything becomes slow to scroll. I loaded another page with 500kb HTML source, opened "Squeezed text" dialog and I couldn't do anything - so slow. Then the dialog crashed with error. I think handling 100kb or even 500kb shouldn't be a problem today on Win 10 and 8 gb machine. How is it possible that such a small thing is causing so many problems for Thonny? I would appreciate if you can fix it asap. Here is the code I am using, feel free to change to another URL to test. 

`
from urllib.request import Request, urlopen

req = Request('http://velenje.com', headers={'User-Agent': 'Mozilla/5.0'})
webpage = urlopen(req).read()

x = len(webpage)
print(x)

print(webpage)
print('*** DONE ***')
`

aivarannamaa, 2020-09-15:
The problem lies in Tk's Text widget, which does not handle long lines properly. That's the reason I implemented text squeezing in the first place.

In your case the HTML page is converted to single string because you are printing a bytes object, not string. Try printing `webpage.decode("UTF-8")` (or whatever encoding the webpage is using).

aivarannamaa, 2020-09-15:
From the content it looks like velenje.com returns data in iso-8859-2 encoding

tiomanisland, 2020-09-15:
>
> The problem lies in Tk's Text widget, which does not handle long lines
> properly. That's the reason I implemented text squeezing in the first place.
>

Could you use some other widget that can handle long text? I mean the text
is not even "long" when it's only 300kb and it is already causing issues.
Can you use some other component?

aivarannamaa, 2020-09-15:
> Can you use some other component?

No

aivarannamaa, 2020-09-15:
Actually, you got me thinking... I could open the squeezed text in another program, eg Notepad in Windows

tiomanisland, 2020-09-16:
That would be a workaround solution but its not that good. Some people
don't mind to open other program each time they run their program - kinda
annoying. The beauty of Thonny is that is is small and fast and you can do
everything inside Thonny. Why you can't use some other "text box"? I mean
anything other Windows programs use? I am not Windows programmer (only PHP)
but I am sure in tool you developed Thonny with there is just few lines to
use some other text box, just to display that long string. What do you
think?

aivarannamaa, 2020-09-16:
Thonny is not a Windows program, it's a cross-platform program (ie. it runs on macOS and Linux besides Windows). Therefore it uses a cross-platform widget toolkit ie. I can't just use a win 32 control.

In principle it is possible to plug in a different platform-specific widget, but it would be too much hassle.

I don't mind leaving this ticket open in case I find more motivation in the future, but don't hold your breath -- there are tens of more important tasks in the list. Most likely Tk 8.7 (where this problem has been fixed) will get released first.

tiomanisland, 2020-09-16:
I understand what you mean. You write one code and compile on different OS?
Still looks strange that this Tk whatever it is has such a problem - I mean
that any component younger than 10 yrs would be unable to handle 1MB. Its a
great tool but if it is really that hard, then one day maybe when you are
free.

aivarannamaa, 2020-09-17:
> unable to handle 1MB

The problem is with long lines only. Did you try printing decoded bytes?

tiomanisland, 2020-09-18:
What does "decoded bytes" mean? I just outputed HTML source from webpage I
fetched with that Python script. I wish I could help...

aivarannamaa, 2020-09-18:
`urlopen(req).read()` returns a `bytes` object, not string. When you print a bytes object, everything gets output as single line, because linebreaks are escaped.

In order to get the string, you should decode it according to the encoding. It looks like the encoding of your example webpage is `iso-8859-2`, therefore instead of doing

```
print(webpage)
```

try doing

```
print(webpage.decode("iso-8859-2"))
```

solsword, 2020-10-06:
Thanks for leaving this issue open; I've got another probable instance of it which I'd like to report in case this changes the priority for this bug. My fellow instructors and I are using Thonny in our intro CS class and we love it, so thanks for putting in the hard work to make this tool! However, in our most recent assignment, we give the students some .py files containing data for them to play with after importing the data into another file. The files are essentially just one long line of text (they were originally .json files) and when our students tried to open them, the reported that Thonny would hang, and one student even reported that after force-closing it, and restarting Thonny, it shows high CPU usage even without that file being open...

I understand that fixing the widget is way out of scope for you right now, but could this problem be detected and an appropriate error issued in the case where a file is being loaded, so that the whole program doesn't crash? Just something like "sorry, Thonny can't display this file because it contains a very long line of text" would be fine for our purposes.

Also, if I can find the time I may try to submit a pull request for this, but I can't promise anything as I've not yet even looked at the Thonny source.

aivarannamaa, 2020-10-21:
@solsword , sorry for late reply. I somehow missed your comment before.

> Just something like "sorry, Thonny can't display this file because it contains a very long line of text" would be fine for our purposes.

I'll consider it.

As I workaround, I recommend you to format this json data into multiple lines.