thonny/thonny #1340: Trying to manage plug-ins restarts computer

Orionsprk, 2020-09-15:
Using Mac OS X Mojave (10.14.6) and trying to access "manage plug-ins" from the Tools menu crashes the computer and restarts it.
Yesterday I was able to invoke the "manage plug-ins" feature but did not install anything. I tried to do the same thing today and just keeps restarting my computer. I tried installing Thonny two more times but the problem is still showing up.
Is there a way I can retrieve and share Thonny logs to review this issue?
thanks for your help on this matter.

aivarannamaa, 2020-09-15:
Thank you for the report!

There should be some logs in `~/Library/Thonny`, but currently they are overwritten on each launch. So, when you experience the problem again then please retrieve the logs before starting Thonny again.

aivarannamaa, 2020-09-22:
Possibly related to #1199 and/or #1188 and https://bugs.python.org/issue34864

aivarannamaa, 2020-09-22:
@Orionsprk , can you please check your "Prefer tabs when opening documents" setting under you Mac's system settings (in section "Dock")? What does it say?

aivarannamaa, 2021-02-27:
Please try whether https://github.com/thonny/thonny/releases/download/v3.3.5/thonny-3.3.5_alt.pkg works better for you. It's with an experimental Universal 2 variant of Python 3.9.2.