thonny/thonny #1341: An unuseful warning

ericd13, 2020-09-19:
Hello

I use Linux Mint 20 with Thonny 3.8.2 (Tk 8.6.10)

Every time I create a graphic with matplotlib.pyplot, I get a warning
qt5ct: using qt5ct plugin
I don't understand the message neither its purpose.

How can I get rid of him ?

aivarannamaa, 2020-09-19:
This doesn't look like Thonny's issue. Please check whether you get the same warning when running your program outside of Thonny (eg. `python3 myscript.py`)

ericd13, 2020-09-19:
Le 19-09-2020 20:07, Aivar Annamaa a écrit :
> This doesn't look like Thonny's issue. Please check whether you get
> the same warning when running your program outside of Thonny (eg.
> python3 myscript.py)

You're right, il appears in other contexts.

Sorry for the inconvenience.

-- 
Eric Détrez
Informatique
Lycée Faidherbe, Lille
Documents de cours en pdf :
faidherbe.org/~informatique

closed 2020-09-20 by aivarannamaa