thonny/thonny #1349: Code completion not working

HobieTCat, 2020-09-23:
I have used Thonny(different versions) on three different Raspberry Pi computers, two version 3B(different raspbian versions), and now a version 4B. Code completion for Thonny has never worked. On version 3 RPi I used IDLE because of this problem and will install it on my new version 4, but since Thonny (3.2.7 is what I have now) is supposed to be something special, it would be nice to get it working.
I saw something about uninstall and reinstall Jedi, which I did. All that did was make the shortcut to Thonny disappear from the Programming menu.
Can you please explain why Code Completion does not work on Thonny?

aivarannamaa, 2020-09-23:
What do you mean by "not working"? Do you get an error message? Do you get wrong completions? Does nothing happen when you invoke code completion? In the latter case, how do you invoke it?

HobieTCat, 2020-09-23:
Not working means that nothing happens, no error, not wrong, just nothing. On most any IDE, when I type the name of something I imported, like numpy, when I type "numpy." a list of available properties/methods pops up and I can select from them. Or, if I type "def mouse_click(" a helper box pops up telling me the arguments that are required. In Thonny, that never happens, but I think it should, I think you call it Code Completion, and when I invoke it, nothing happens. That's what I mean by "not working".

aivarannamaa, 2020-09-23:
Thonny doesn't pop up code completion box automatically (as many users don't want it). You need to invoke it with Ctrl+Space.

HobieTCat, 2020-09-23:
Thanks, I didn't know that. Is it supposed to have the helper box part too? Telling the arguments required?

aivarannamaa, 2020-10-05:
Sorry for late reply

> Is it supposed to have the helper box part too? Telling the arguments required?

No, at the moment Thonny doesn't support this.

closed 2020-10-05 by aivarannamaa

jcinternetstuff, 2023-10-27:
Code Completion still does not work on Thonny 4.1.3.

And have "ctrl.space" to initiate it when a simple "do"' can do this, like in most other IDEs?