thonny/thonny #1350: Device message ! MicroPython commands have been moved or replaced

FreddyBI, 2020-09-23:
[image]

and shell
>>> %lsdevice
Unknown command: lsdevice

[image]
[image]
No problem to save, save as ... or run my scripts or open files from MicroPython device.
[image]

aivarannamaa, 2020-09-23:
This Thonny version doesn't have `%lsdevice` command anymore. Use View => Files instead.

FreddyBI, 2020-09-23:
Remove "Device" from menu bar "File Edit View ...." ?
Or remove Message on click Device ?

aivarannamaa, 2020-09-23:
Sorry, I did not understand what is the problem.

FreddyBI, 2020-09-23:
I use for the first time Thonny program.
When  I click to Device on the menu bar I receive info moved commando's.
This is good info for users that use already previously versions  Thonny, not for me.

aivarannamaa, 2020-09-23:
> I use for the first time Thonny program.
> When I click to Device on the menu bar I receive info moved commando's.
> This is good info for users that use already previously versions Thonny, not for me.

I think it's quite common that some parts of a program are not useful for all users. Do you propose I should remove this menu?

FreddyBI, 2020-09-23:
Yes.
(I like use Thonny, thanks)

closed 2020-09-24 by FreddyBI