thonny/thonny #1352: Mysql connectivity

SSGowri, 2020-09-29:
Hi,
I can't do Mysql connectivity in  thonny. in manage packages mysql is not listing to install. Please tell me a solution

aivarannamaa, 2020-09-29:
If you want your code to be able to talk to a MySQL server, then you need to select a suitable Python library for this and install it (eg. via Thonny's package manager). The choice of the library is not Thonny-specific.

closed 2020-09-29 by aivarannamaa

SSGowri, 2020-09-29:
I tried to install mysql. but its not showing in manage packages. Screen shot is attached. 
Regards ,

Gowri S S

Screen shot    Save Paper ! Think Before You Print !!

SSGowri, 2020-09-29:
I tried this way also(screen shot attached). but error msg only displaying
Regards ,

Gowri S S

    Save Paper ! Think Before You Print !!

 
 

Closed #1352.

aivarannamaa, 2020-09-30:
Attachments don't come through if you reply by e-mail. You can post screenshots only on GitHub interface. 

Are you sure you want `mysql` PyPI package? Or are you trying to install MySQL server via Thonny?

SSGowri, 2020-09-30:
I tried to install Mysql via Thonny for doing connectivity. but i cant install mysql errors are coming. screen shot attached.
Please do the needful.
Regards ,

Gowri S S

    Save Paper ! Think Before You Print !!

aivarannamaa, 2020-09-30:
As I told you, I don't see your screenshots if you are replying by e-mail.

SSGowri, 2020-09-30:
**here also i included**
Please do the needfull

[image]
[image]

aivarannamaa, 2020-09-30:
Sorry, but this is not Thonny's problem. If you can install this package with plain `pip install`, but not with Thonny's package manager, then I'll investigate further.

r0the, 2020-09-30:
According to PyPI, mysql is an alias for MySQL-python, which is Python 2.7 only. It will not work with Thonny and Python 3.

I'd use the official mySQL driver for Python: [mysql-connector-python](https://pypi.org/project/mysql-connector-python/). It installs nicely in Thonny.

SSGowri, 2020-09-30:
i went through that page .i am not getting from where to install from that page

SSGowri, 2020-09-30:
[image]

SSGowri, 2020-09-30:
error is comming

SSGowri, 2020-09-30:
can you please send me a video showing next steps

aivarannamaa, 2020-09-30:
@SSGowri, I recommend you ask advice from a Python forum. Your problem is not specific to Thonny. Also, I recommend to postpone MySQL for now and learn basic Python programming first.

SSGowri, 2020-09-30:
I know python programming. I am a teacher . my collegue told about Thonny. and i taught my students using tonny. next topic is connectivity. thats why i am trying this