thonny/thonny #1353: Opening on the last file

laboiteaphysique, 2020-09-30:
Hi,
I let unchecked the option "Reopen all files from previous session", but Thonny allways open the last file used.
Could you add an other checkbox in these options tab like "Don't open the last file on start"
(I'm a teacher and my students use the same session. When opening Thonny, the next student see the file opened by the other one, and sometime reuse it...).
Best regards
Vincent Le Mieux

aivarannamaa, 2020-09-30:
Thank you for the suggestion!

I'll consider changing the binary option to a choice between ["all", "last", "none"]

laboiteaphysique, 2020-09-30:
Thank for your positive answer !
... waiting for a new release yet !
Best regards
Vincent Le Mieux

eyleenk, 2023-10-16:
still having the same issue, I've overwritten my other files because of this

aivarannamaa, 2024-05-30:
> I'll consider changing the binary option to a choice between ["all", "last", "none"]

I changed my mind. I introduced new option "Reopen files from previous session" instead of "Reopen **all** files from previous session". It is checked by default, but if it is unchecked, no files from previous session get opened automatically.

closed 2024-05-30 by aivarannamaa

amathenge, 2024-11-26:
Hi - the option does not seem to work on Thonny for Mac.
I installed Thonny via homebrew, and works excellent.

However, when I launch it, the old open files are always open even if I clear the checkbox for that option

[image]

Thanks,
Andrew.

aivarannamaa, 2025-01-15:
The new behavior will be published as Thonny 5.0