thonny/thonny #1355: STM32 large file writes  (> 2kB) failing

curtywang, 2020-10-02:
### Problem
Due to the SRAM size of some STM32 (namely STM32F4xx) microcontrollers, MicroPython occupies 92/96 kB of SRAM.
This makes it so when uploading files using the Upload feature, there will be unpredictable behavior with the write (either the write will overwrite other files in SRAM or simply fail and cause timeouts in Thonny).
This same problem existed previously in rshell: https://forum.micropython.org/viewtopic.php?t=6567

### Solution
dhyland's solution was to add a call to os.sync() during the block writes.
I have verified by adding "import os" to the setup calls inside _write_file_via_serial() in bare_metal_backend, along with the following call after the block write:
```
hex_mode = self._should_hexlify(target_path)
if hex_mode:
    self._execute_without_output(
        dedent(
            """
        from binascii import unhexlify as __thonny_unhex
        import os  # new code
        def __W(x):
            global __thonny_written
            __thonny_written += __thonny_fp.write(__thonny_unhex(x))
            __thonny_fp.flush()
    """
        )
    )
else:
    self._execute_without_output(
        dedent(
            """
        import os  # new code
        def __W(x):
            global __thonny_written
            __thonny_written += __thonny_fp.write(x)
    """
        )
    )

...

bytes_sent += len(block)  # pre-existing code to show location
out, err = self._execute("if hasattr(os, 'sync'):\n    os.sync()", capture_output=True)  # new code for fix
```

I can also submit a pull request if you'd like.

aivarannamaa, 2020-10-05:
Thank you for this suggestion!

Thonny used to use `os.sync` when writing files, but I removed it after I added `f.flush()`, because I really didn't understand what `os.sync` does. 

I could not reproduce the problem from current master (maybe because I have an old MicroPython version on my Nucleo F401RE), but I'm happy to add it back if you say it fixes your case (especially as it seems to not hurt performance).

I made the changes myself but I really appreciate you thorough report and diagnosis!

curtywang, 2020-10-06:
OK, thank you!  It is likely in how the STM32 driver handles writing to the filesystem... I'm imagining f.flush() flushes to file but does not necessarily order the filesystem to commit to disk.  I'm not sure on the details myself but if dhylands had to add it to rshell then I guess it is a micropython thing.

closed 2020-10-06 by aivarannamaa