thonny/thonny #1357: Thonny breaks when emojis are used

BioHazard1282, 2020-10-05:
I am making an example program in Python 3 for a video, but Thonny 3.2.7 breaks when any line has emojis in it:
`def _thet():`
`    return "I am his brother. He is a memer, I am a coder."`
`    return "Favorite Emojis: 🦅🐍🦎🐊🐉🐲`
`print(_thet())`

If you were to paste this in an editor, another "Favorite Emojis" text appears at the end but white outside of comments, and I have to force-quit when I do this. I am using MacOS Mojave.

aivarannamaa, 2020-10-06:
Thank you for the report!

It is a known issue and I'm waiting for newer Tk to be bundled with Python: #32

closed 2020-10-06 by aivarannamaa

AgimaFR, 2022-04-01:
In the meantime, you can use one of the following workarounds for source code management in Thonny and **run the script in a terminal window**(not in Thonny).

1. Use the [Unicode code of emojis](https://unicode.org/emoji/charts/emoji-list.html)
```python
print("\U0001F1EC\U0001F1E7")
print("\U0001F923")
```
2.Use the library [emoji](https://pypi.org/project/emoji2/)
```python
import emoji
print(emoji.emojize(":grinning_face_with_big_eyes:"))
print(emoji.emojize(":winking_face_with_tongue:"))
print(emoji.emojize(":zipper-mouth_face:"))
```