thonny/thonny #1360: Problem with Thonny

tomasdaureli, 2020-10-08:
Hi i'm Tomas. i had a problem with my Thonny. I think that I put the mode expert and now the screen looks very small and i can't see anything. Please i need help because i'm student. I don't know how fix it. 
[image]

aivarannamaa, 2020-10-08:
Please go to "Tools => Options => General". What do you have in "UI scaling factor" box? Try larger values. You need to restart Thonny after changing this.

aivarannamaa, 2020-10-08:
You may need to change "Font scaling" mode as well

tomasdaureli, 2020-10-08:
I can’t find the option “UI scalling factor” box. The tools looks like this. I tried to change the font scalling too, but it doesn’t change the rest of the view.     From: Aivar AnnamaaSent: jueves, 8 de octubre de 2020 02:17To: thonny/thonnyCc: tomasdaureli; AuthorSubject: Re: [thonny/thonny] Problem with Thonny (#1360) Please go to "Tools => Options => General". What do you have in "UI scaling factor" box? Try larger values. You need to restart Thonny after changing this.

tomasdaureli, 2020-10-08:
I uninstall the application too, but the configuration is still the same and nothing is changed. Is there any way to restore the app to the original values, installing from zero? Thank you   From: Aivar AnnamaaSent: jueves, 8 de octubre de 2020 02:18To: thonny/thonnyCc: tomasdaureli; AuthorSubject: Re: [thonny/thonny] Problem with Thonny (#1360) You may need to change "Font scaling" mode as well

aivarannamaa, 2020-10-08:
For resetting the configuration:

* Start Thonny
* Select "Tools => Open Thonny data folder"
* Close Thonny
* delete configuration.ini
* Start Thonny

tomasdaureli, 2020-10-08:
Thank you very much!!!!! You just saved my life. Greetings.   From: Aivar AnnamaaSent: jueves, 8 de octubre de 2020 16:11To: thonny/thonnyCc: tomasdaureli; AuthorSubject: Re: [thonny/thonny] Problem with Thonny (#1360) For resetting the configuration:Start ThonnySelect "Tools => Open Thonny data folder"Close Thonnydelete configuration.iniStart Thonny

closed 2020-10-08 by tomasdaureli

Flaming-crossbow, 2020-12-06:
> * configuration.ini

Thank you soo frggn much,