thonny/thonny #1363: import numpy in shell view puased at startup . (version 3.3.0b3)

beardad1975, 2020-10-11:
I  cloned thonny version 3.3.0b3 and found  a strange thing in shell view.

When I tried to import numpy in shell view, it will paused . 
I needed to press  an extra Enter key to get  it run .

My simple numpy program do  "Run in current Script"  will pause at startup and needs an an extra Enter.
But  the same program do "Run current script in terminal"  just normally.

By the way .   My laptop os is Win10 64bit. 

[image]

aivarannamaa, 2020-10-11:
Thank you for the report!

I was able to reproduce this. It looks like I have to revert a design decision because it distracts DLL loading machinery.

aivarannamaa, 2020-10-11:
I think I was able to fix this problem. Please pull the updates and see if this works for you!

beardad1975, 2020-10-12:
I tried the updates.  It worked  as usual .  Good job.

closed 2020-10-12 by beardad1975