thonny/thonny #1364: Show full path+name when mouse points(hovers) editor tab

IhorNehrutsa, 2020-10-12:
Feature request.
I lost some time when I opened one module from another drive.
The hint under the mouse cursor (see a picture) will help to resolve my misunderstanding.
Thanks a lot.
[image]

Thonny 3.3.0b2 used

aivarannamaa, 2020-10-21:
I'll consider adding it, but did you notice that the title bar of Thonny's window has full path of the active file?

IhorNehrutsa, 2020-10-21:
Not always.
[image]

aivarannamaa, 2020-10-21:
Oh, this is a bug and must be fixed.

IhorNehrutsa, 2020-10-21:
Not the big bug, after F5(Run) correct full name is displayed.

aivarannamaa, 2020-10-28:
Can't reproduce mismatching titlebar path in Linux. Will test in Windows later.

aivarannamaa, 2023-08-20:
Will be reimplemented with new Notebook

closed 2023-08-20 by aivarannamaa