thonny/thonny #1365: Remote Python 3 (SSH) interpreter error message

Standaros, 2020-10-12:
I have a problem in Thonny 3.3.0b2 (Windows version), when i try to configure remote Python interpreter over ssh to my linux machine session is opened fine but immediately ends with error message :

/usr/bin/python3: No module named thonny.plugins.cpython.__main__; 'thonny.plugins.cpython' is a package and cannot be directly executed

Backend terminated or disconnected. Use 'Stop/Restart' to restart.

Standaros, 2020-10-13:
Ok i have figured it out, you need to have Thonny 3.3.0b2 on both sides of ssh connections, on your Win machine and on Liux server. Then SSH connection to remote interpreter works

aivarannamaa, 2020-10-14:
Thank you for the report!

Actually my plan was not to require Thonny in the remote computer at all -- the local Thonny is supposed to upload required modules to a temp folder in the beginning of the session.

I'll try to reproduce your case.

aivarannamaa, 2020-10-20:
I could not reproduce this. Could you please try this again with the latest beta (https://github.com/thonny/thonny/issues/1377#issuecomment-712640252)?

Is `/tmp/thonny-backend-3.3.0b6/thonny/plugins/cpython/__main__.py` present in your remote machine?

Standaros, 2020-10-21:
I have tried latest version as you suggested,  __main_.py is present, __init__.py is not.

Problem for me is still same when starting from scratch, but can be fixed quite easilly i believe :
 on local and on remote machine i have Thonny version: 3.3.0b6, when connecting from win to linux(raspbian) there was this error message  :

```
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/tmp/thonny-backend-3.3.0b6/thonny/plugins/cpython/__main__.py", line 38, in <module>
    thonny.configure_backend_logging()
  File "/tmp/thonny-backend-3.3.0b6/thonny/__init__.py", line 306, in configure_backend_logging
    _configure_logging("backend.log", logging.WARNING)
  File "/tmp/thonny-backend-3.3.0b6/thonny/__init__.py", line 318, in _configure_logging
    file_handler = logging.FileHandler(log_file, encoding="UTF-8", mode="w")
  File "/usr/lib/python3.7/logging/__init__.py", line 1092, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python3.7/logging/__init__.py", line 1121, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding)
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/.config/Thonny/backend.log'
```

So i have tried to create empty directory on remote machine - /home/pi/.config/Thonny/ , tried to reconnect and session to remote interpreter is established fine !  
Im not sure why this directory was not created automatically, all i did before updating to dev version was uninstall command to remove current version of Thonny included in OS distribution(apt-get remove thonny)

backend.log file that is created in this directory containst only one line :
`INFO    thonny: Thonny version: 0.0.0`

Thank you for your work on this amazing piece of software

aivarannamaa, 2020-10-21:
Weird, I found this missing config-dir problem in b4 and [fixed in b5](https://github.com/thonny/thonny/commit/889f9584aade17909231fe60213347a634ad252b) and I can't reproduce in b6.

> backend.log file that is created in this directory containst only one line :
> INFO thonny: Thonny version: 0.0.0

Thanks for noticing this -- I should also upload the VERSION file.

I'll test this thing more thoroughly tomorrow.

Standaros, 2020-10-22:
To be sure i did same process with Ubuntu 20.04 64bit and error is same, when connecting to remote  machine

```
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/tmp/thonny-backend-3.3.0b6/thonny/plugins/cpython/__main__.py", line 38, in <module>
    thonny.configure_backend_logging()
  File "/tmp/thonny-backend-3.3.0b6/thonny/__init__.py", line 306, in configure_backend_logging
    _configure_logging("backend.log", logging.WARNING)
  File "/tmp/thonny-backend-3.3.0b6/thonny/__init__.py", line 318, in _configure_logging
    file_handler = logging.FileHandler(log_file, encoding="UTF-8", mode="w")
  File "/usr/lib/python3.8/logging/__init__.py", line 1143, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python3.8/logging/__init__.py", line 1172, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding)
FileNotFoundError: [Errno 2] No such file or directory: '/home/ubuntu/.config/Thonny/backend.log'
```
If i create directory Thonny in user home directory under .config then session is opened fine

aivarannamaa, 2020-11-11:
Should be fixed now. Fix will be in the final 3.3.0 in couple of days.

closed 2020-11-11 by aivarannamaa