thonny/thonny #1367: Request for a layout change option

pssolanki111, 2020-10-14:
First up, I loved the software. It's really great.

Just a suggestion/request though which should be pretty easy for you as it's just related to layout.

I'm on a windows machine amd as the thonny software shows the editor window and shell window vertically adjacent. I mean shell window is below the editor window which is fine. 
I actually would love to be able to use those windows side by side. I mean horizontally adjacent. Simply said, I want the shell window to be where the assistant window usually occurs. And of course if we can rearrange the layout then nothing better than that.

I actually use thonny for competitive programming so in the current layout, I have to scroll a lot in both sections to get somewhere. 

Again thank you for this amazing software, and I hope you'll consider my request

- P S Solanki

aivarannamaa, 2020-10-14:
Thank you for the kind words!

This feature is already in my TODO list (#544). Here is a workaround: https://github.com/thonny/thonny/wiki/Custom-layout

closed 2020-10-14 by aivarannamaa

pssolanki111, 2020-10-14:
Amazing!! The workaround works perfect. even though I'll wait for the feature to be added, i am totally fine as of now. I just needed a good alternate of IDLE for competitive. I already have PyCharm, sublime and notepad++ and thonny is now one of my favs.

Great work, Aivar....