thonny/thonny #1368: unable to install numpy

ThibaudTC, 2020-10-14:
trying to install numpy I ge an error:
ERROR: Command errored out with exit status 1: 'P:\Python39\python.exe' 'P:\Python39\lib\site-packages\pip\_vendor\pep517\_in_process.py' prepare_metadata_for_build_wheel 'C:\Users\Public\Documents\Wondershare\CreatorTemp\tmpgmx02eyg' Check the logs for full command output.
Process failed, return code: 1

what am I doing wrong? using Thonny 3.2.7

aivarannamaa, 2020-10-14:
Please try "Tools => Open system shell" and run `pip install --user numpy`. Do you get the same error?

What is `C:\Users\Public\Documents\Wondershare\CreatorTemp\`? Your current directory?

ThibaudTC, 2020-10-14:
> <https://github.com/thonny/thonny/issues/1368#issuecomment-708303668>view
> it on GitHub, or

Numpy is now instralled but I get several errors 
in _inint_ and others. Is numpy ready for python 3.9?

The line with 
C:\Users\Public\Documents\Wondershare\CreatorTemp\? 
Your current directory? is also strange for me

aivarannamaa, 2020-10-14:
Numpy source code may be ready, but it looks like they haven't published binary wheels for 3.9: https://pypi.org/project/numpy/1.19.2/#files

I wouldn't bother building from source on Windows. I recommend you use latest 3.8 for now.

closed 2020-10-14 by aivarannamaa

ThibaudTC, 2020-10-16:
i stay now on python 3.8.
Thanks