thonny/thonny #1369: Problem Thonny´s backend - Timeout ESP32 Wroom

xal88, 2020-10-14:
Hi,
thanks for your great IDE. 
having sporadic  disconnects ( chance 10% )  when running shell command. 
Happens more often when files are larger ( but not huge )
ESP gc.mem_free() > 20k
Disconnects do not happen when running via Putty serial.

Thonny 3.2.7, MicroPython v1.13 on 2020-09-02;

```
<thonny>{'uos': "<module 'uos'>", 'run_wheel': '<function run_wheel at 0x3ffe9050>', 'set_all': '<function set_all at 0x3ffe8a20>', 'bdev': '<Partition type=1, subtype=129, address=2097152, size=2097152, label=vfs, encrypted=0>', 'shift_left': '<function shift_left at 0x3ffe8fa0>', 't1': '[[16, 16, 16], [16, 16, 16], [16, 16, 16], [16, 16, 16], [16, 16, 16], [16, 16, 16], [16, 16, 16], [16, 16, 16]]', 'network': "<module 'network'>", 'micropython': "<module 'micropython'>", 'wheel': '<function wheel at 0x3ffe9120>', 't2': '[[4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4], [4, 4, 4, 4]]', 'fade_out': '<function fade_out at 0x3ffe8f40>', 's2': '<NeoPixel object at 3ffe5670>', 'connectWifi': '<function connectWifi at 0x3ffe4e20>', 'rainbow_cycle': '<function rainbow_cycle at 0x3ffe8f80>', 's1': '<NeoPixel object at 3ffe5890>', 'utime': "<module 'utime'>", 'wipe': '<function wipe at 0x3ffe9000>', 'blend': '<function blend at 0x3ffe8f60>', 'set_colors': '<function set_colors at 0x3ffe5bb0>', 'wheel1': '<function wheel1 at 0x3ffe9140>', 'shift_right': '<function shift_right at 0x3ffe8fd0>', 'pixel_maxpower': '128', 'wheel2': '<function wheel2 at 0x3ffe9020>', 'set_target': '<function set_target at 0x3ffe5c20>', 'gc': "<module 'gc'>", 'machine': "<module 'umachine'>", 'esp': "<module 'esp'>", 'time': "<module 'utime'>", 'neopixel': "<module 'neopixel' from 'neopixel.py'>", 'dimm': '<function dimm at 0x3ffe8a40>', 'fade_in': '<function fade_in at 0x3ffe8f10>'}</thonny>>
```

```
PROBLEM WITH THONNY'S BACK-END:

Traceback (most recent call last):
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 35, in read
    self._read_buffer.extend(self._read_queue.get(True, timer.time_left))
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\queue.py", line 178, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 340, in handle_command
    response = handler(cmd)
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 712, in _cmd_get_globals
    "{name : repr(value) for (name, value) in globals().items() if not name.startswith('__')}"
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 468, in _evaluate
    _, _, value_repr = self._execute_print_expr(expr, prelude, cleanup)
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 465, in _execute_print_expr
    return self._execute(script, capture_output)
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 440, in _execute
    return self._process_until_raw_prompt(capture_output)
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 531, in _process_until_raw_prompt
    self._check_for_side_commands()
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 634, in _check_for_side_commands
    self._submit_input(cmd.data)
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 389, in _submit_input
    echo = self._connection.read(len(bdata))
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 37, in read
    raise TimeoutError("Reaction timeout. Bytes read: %s" % self._read_buffer)
TimeoutError: Reaction timeout. Bytes read: bytearray(b'')
Crash in backend
Traceback (most recent call last):
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 35, in read
    self._read_buffer.extend(self._read_queue.get(True, timer.time_left))
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\queue.py", line 178, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 104, in __init__
    self._mainloop()
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 140, in _mainloop
    self._submit_input(cmd.data)
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 389, in _submit_input
    echo = self._connection.read(len(bdata))
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 37, in read
    raise TimeoutError("Reaction timeout. Bytes read: %s" % self._read_buffer)
TimeoutError: Reaction timeout. Bytes read: bytearray(b'')
Traceback (most recent call last):
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 35, in read
    self._read_buffer.extend(self._read_queue.get(True, timer.time_left))
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\queue.py", line 178, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 104, in __init__
    self._mainloop()
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 140, in _mainloop
    self._submit_input(cmd.data)
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 389, in _submit_input
    echo = self._connection.read(len(bdata))
  File "C:\Users\as\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 37, in read
    raise TimeoutError("Reaction timeout. Bytes read: %s" % self._read_buffer)
TimeoutError: Reaction timeout. Bytes read: bytearray(b'')

Backend terminated or disconnected. Use 'Stop/Restart' to restart.
```

aivarannamaa, 2020-10-14:
Please check whether https://github.com/thonny/thonny/releases/tag/v3.3.0b2 works better for you!

xal88, 2020-10-14:
Hello,
installed the 3.3.03b, and worked wuth it for 3 hours - no issues so far.
I will close this, and reopen if needed.
Thanks and kind regards

closed 2020-10-14 by xal88