thonny/thonny #1370: STM32 Nucleo and Micropython - save files on flash

alecl84, 2020-10-14:
I have problems in saving Micropython files on STM32 Nucleo (model F767ZI) flash memory (save copy-> MicroPython device)
Small files (few lines of code) give no problems, medium files (up to 35 lines of code) sometimes fail, large files (more than 40 lines of code) fail all the time. I tried with version 3.2.6 and 3.3.0b2, both on Windows and Raspberry versions, same effect.
I tried also with short and long file names, no effect.
If i run the script by pressing F5 key or the green arrow icon, the file is correctly processed and executed, but it is not stored permanently on board memory so every time the system is powered off everything is ereased.

here below the error messages I received:

3.3.0b2

MicroPython v1.12-452-g801f7dca7 on 2020-05-14; NUCLEO-F767ZI with STM32F767

PROBLEM WITH THONNY'S BACK-END:
Could not read command confirmation. Got b''

SCRIPT:
__W(b"are(half_per+angolo//2)\r\n        timA_CH2.compare(half_per-angolo//2)\r\n    else:#CH2 in ritardo su CH1\r\n        #problemi di shift angolo, inverte l'onda per cambi repentini con angolo prossimo a 90\xc2\xb0\r\n        #SOLUZIONE: ritardare CH2 e anticipare CH1 di met\xc3\xa0 angolo (//2, divisione senza resto), rispettivamente\r\n        timA_CH1.compare(half_per+angolo//2)\r\n        timA_CH2.compare(half_per-angolo//2)\r\n\r\ndef cycle(timer):\r\n    global k\r\n    k=True\r\n\r\ntimA_CH1=timA.channel(1, pyb.Timer.OC_TOGGLE, compare=h")

3.2.6

MicroPython v1.12-452-g801f7dca7 on 2020-05-14; NUCLEO-F767ZI with STM32F767

PROBLEM WITH THONNY'S BACK-END:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/connection.py", line 35, in read
    self._read_buffer.extend(self._read_queue.get(True, timer.time_left))
  File "/usr/lib/python3.7/queue.py", line 178, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 340, in handle_command
    response = handler(cmd)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 743, in _cmd_write_file
    self._write_file(generate_blocks(cmd["content_bytes"], BUFFER_SIZE), cmd["path"])
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 1094, in _write_file
    result = self._write_file_via_serial(content_blocks, target_path, notifier)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 1166, in _write_file_via_serial
    self._execute_without_output(script)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 444, in _execute_without_output
    out, err, value = self._execute(script, capture_output=True)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 438, in _execute
    ok = self._connection.read(2)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/connection.py", line 37, in read
    raise TimeoutError("Reaction timeout. Bytes read: %s" % self._read_buffer)
TimeoutError: Reaction timeout. Bytes read: bytearray(b'')

aivarannamaa, 2020-10-14:
It looks like #1355, ie. if the fix should be in the next version.

If you are willing to [run Thonny from the source](https://github.com/thonny/thonny/wiki/Running-from-the-source), you could check, whether current master fixes your case as well.

alecl84, 2020-10-15:
It seems also similar to #936, that is older, about one year ago. I tried to read all the discussion, but honestly I did not undestand if that issue has been solved and in case how.

aivarannamaa, 2020-10-15:
@alecl84, the discussion under #936 is obsolete when you are using Thonny 3.3.

aivarannamaa, 2021-01-16:
Please try newest Thonny version and reopen if you still have this problem.

closed 2021-01-16 by aivarannamaa