thonny/thonny #1377: Simple way of installing new beta?

speedy-10, 2020-10-19:
Hi, exists a simple method to install the new beta, something like  

- bash <(wget -O - https://thonny.org/installer-for-linux-beta)
- .deb File ?

That would be helpful for me.

Regards  Speedy-10

Standaros, 2020-10-20:
Hi,
 if you want to install pre-release or beta version there is page where Aivar is publishing builded packages - check here https://github.com/thonny/thonny/releases , latest is from 3rd Sept with Version 3.3.0b2, but expect some bugs as its under heavy developlent

if you want to have current version you can use git clone command to clone current status of repository and then build it yourself, its not really hard to build - if you want sent me DM and i can try to help you with building. I am currently doing this on Windows and Linux(debian) machines that im using

aivarannamaa, 2020-10-20:
Here are commands for Linux, if you want to test latest betas (without published binaries):

```
python3 -m venv thonny_venv
source thonny_venv/bin/activate
pip install --pre --no-cache-dir -U thonny
thonny
```

speedy-10, 2020-10-20:
thanks.
Can I ignore this warnings?
`Collecting wrapt~=1.11
  Downloading wrapt-1.12.1.tar.gz (27 kB)
Building wheels for collected packages: wrapt
  Building wheel for wrapt (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/matthias/thonny_venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-6vrd7deu/wrapt/setup.py'"'"'; __file__='"'"'/tmp/pip-install-6vrd7deu/wrapt/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-_uvvsck7
       cwd: /tmp/pip-install-6vrd7deu/wrapt/
  Complete output (6 lines):
  usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: setup.py --help [cmd1 cmd2 ...]
     or: setup.py --help-commands
     or: setup.py cmd --help
  
  error: invalid command 'bdist_wheel'
  ----------------------------------------
  ERROR: Failed building wheel for wrapt
  Running setup.py clean for wrapt
Failed to build wrapt
ERROR: jedi 0.17.2 has requirement parso<0.8.0,>=0.7.0, but you'll have parso 0.8.0 which is incompatible.
Installing collected packages: six, asttokens, lazy-object-proxy, wrapt, astroid, mccabe, isort, toml, pylint, mypy-extensions, typing-extensions, typed-ast, mypy, pyserial, docutils, Send2Trash, parso, jedi, thonny
    Running setup.py install for wrapt ... done
Successfully installed Send2Trash-1.6.0b1 astroid-2.4.2 asttokens-2.0.4 docutils-0.16 isort-5.6.4 jedi-0.17.2 lazy-object-proxy-1.4.3 mccabe-0.6.1 mypy-0.790 mypy-extensions-0.4.3 parso-0.8.0 pylint-2.6.0 pyserial-3.5b0 six-1.15.0 thonny-3.3.0b6 toml-0.10.1 typed-ast-1.4.1 typing-extensions-3.7.4.3 wrapt-1.12.1

`

aivarannamaa, 2020-10-20:
The wheel problem can be ignored, but jedi vs parso conflict may affect code completion. I'll try to fix this in the next beta.

speedy-10, 2020-10-20:
>aivarannamaa

Thanks for your help with the commands, very comfortable to test beta versions.

closed 2020-10-20 by speedy-10