thonny/thonny #1378: I can't install pandas

caroblees, 2020-10-19:
I need to install pandas, but I get this ad, which I already tried to solve but it did not work
[image]

aivarannamaa, 2020-10-20:
I see two problems:

* In terminal you are using Python 3.9, but it looks like Python 3.9 binary wheels for Pandas (and numpy) are not published yet to PyPI, and building such complex packages from source is error prone.
* If you manage to install Pandas to your Python 3.9, you need to make sure Thonny is using this interpreter (Run => Select interpreter). By default it is using bundled Python 3.7 (with its own packages).

aivarannamaa, 2020-10-20:
It's quite common that binary wheels for new Python version don't get published before first bugfix release (eg. 3.9.1)

closed 2021-01-16 by aivarannamaa