thonny/thonny #1379: MatPlotLib Graphics

Quimel1, 2020-10-20:
Hi, I'm new with this, so I hope I explain myself correctly. 
I'm learning Python in my school, and we are using libraries as MatPlotLib. 
You'll know that when you visualize a graphic, it opens another window. So today I'm having a final presentation, but while I was testing my program just to check everything was in order I was just wandering trough the Thonny Options, and I have no idea of what I clicked-on, but now each time I try to visualize a Graphic the computer kind of resets. 
I use a MacBook Pro and Thonny to run my code. 
I will appreciate any help.

aivarannamaa, 2020-10-20:
If you experience this crash again (or if you haven't started Thonny since last crash) then please check whether Thonny logged something: https://github.com/thonny/thonny/wiki/Troubleshooting#see-the-logs-macos

If there is anything, then please post them here.

Also, in the same directory as logs, there is a file configuration.ini, which may also be useful in debugging this situation.

Quimel1, 2020-10-20:
I entered /Library/Thonny and inside the folder there are 6 files (as the image shows), but I only could upload one of them, because the page says the file is empty, and for the .ini file it says the page doesn't support this type of file (I am ommiting user_notes.txt)

 

[image]

[frontend.log](https://github.com/thonny/thonny/files/5411978/frontend.log)

aivarannamaa, 2020-10-21:
Unfortunately there was nothing useful in frontend.log.

Could you please make a copy of configuration.ini, rename it to have `.txt` suffix and then send it here?

Quimel1, 2020-10-24:
[configuration.txt](https://github.com/thonny/thonny/files/5432395/configuration.txt)

Hi, this is configuration.ini converted to .txt

aivarannamaa, 2020-10-24:
Unfortunately I couldn't spot anything suspicious in the configuration.ini

> now each time I try to visualize a Graphic the computer kind of resets

Can you please be more specific? Do you get logged out from your session?

There couple of more things to try:

- Close Thonny and delete your configuration.ini. After this your Thonny will open with default settings.
- Try the beta of the next version: https://github.com/thonny/thonny/releases/tag/v3.3.0b2

-

aivarannamaa, 2020-10-24:
If none of these help, then please post an example program, which reproduces this problem for you.

closed 2021-01-16 by aivarannamaa