thonny/thonny #1383: esp32 incompatibility problem

gooyle, 2020-10-22:
sometimes thonny may lose connection with esp32，and error is about : `<thonny>{'/': {'boot.py': {'size': 139, 'kind': 'file', 'time': 946684800}, 'main.py': {'size': 200, 'kind': 'file', 'time': 946685163}}}</thonny>>`

aivarannamaa, 2020-10-22:
Please try whether https://github.com/thonny/thonny/releases/tag/v3.3.0b2 works better for you!

grubert, 2020-10-23:
> sometimes thonny may lose connection with esp32，and error is about : <thonny>{'/':
> {'boot.py': {'size': 139, 'kind': 'file', 'time': 946684800}, 'main.py':
> {'size': 200, 'kind': 'file', 'time': 946685163}}}</thonny>>
i tried 3.2.7 on linux, macosx and several windows pcs.
linux and mac work but on windows the file list (see above) always ends up
in the python console
and not in the file view.

3.3.0b2 works on all systems
upgraded with pip install --pre thonny

aivarannamaa, 2020-10-24:
@grubert, thanks, that's good to know!

Closing now. @gooyle, please reopen if the problem persists with version 3.3

closed 2020-10-24 by aivarannamaa

gooyle, 2020-10-25:
> Closing now. @gooyle, please reopen if the problem persists with version 3.3

thx，the 3.3 version works.