thonny/thonny #1384: Can we please have a way to move shell to the right instead of below the editor?

youstina27, 2020-10-24:


aivarannamaa, 2020-10-24:
This is in my TODO list: #544.

aivarannamaa, 2020-10-24:
And here is the workaround: https://github.com/thonny/thonny/wiki/Custom-layout

closed 2020-10-24 by aivarannamaa

MattTheBanana, 2022-10-24:
This workaround doesn't work in the latest version. I tried the workaround and it simply resets when you open the program.

aivarannamaa, 2022-10-29:
@MattTheBanana, did you close Thonny before saving the configuration file? Otherwise Thonny will rewrite your changes on close.

MattTheBanana, 2022-10-30:
Yes, I closed it and saved it before opening Thonny at all.

aivarannamaa, 2022-12-28:
@MattTheBanana, I can't reproduce the problem in Thonny 4.0.1.

MattTheBanana, 2022-12-28:
<iframe src="https://www.veed.io/embed/420eaf15-d25e-40f3-ad4b-dfa226fe9564" width="744" height="504" frameborder="0" title="rec_7.mp4" webkitallowfullscreen mozallowfullscreen allowfullscreen></iframe>

aivarannamaa, 2022-12-29:
@MattTheBanana, thank you for the video! I saw 2 issues:

* You seemed to save the configuration first and then restart Thonny. This doesn't work. You need to close Thonny, then save the configuration and then start Thonny
* You changed wrong parameters in the configuration file. Pay attention to the point 4 at https://github.com/thonny/thonny/wiki/Custom-layout

MattTheBanana, 2022-12-29:
Hi, sorry for the confusion. I got it to work. I thought the page was outdated when I saw multiple places for the parameter, but I shouldn't have assumed.
Thank you for the help!