thonny/thonny #1385: More Large File Upload Issue

jackho9, 2020-10-25:
Hi, thank you for thonny!

I am running thonny 3.2.7 on windows 10, talking via USB to an M5stickC Plus, which is based on a ESP32-Pico-D4 with 520kB of internal Ram.  gc.mem_free() returns about 80kB of available ram when first reset with the firmware as set up:

`MicroPython 43c9ed5e1-dirty on 2020-09-30; M5StickC-Plus with ESP32

Type "help()" for more information.
>>> import gc
>>> gc.mem_free()
81936`

I can run my 4kB program line by line via the command line but when I try to upload it to the Stick thonny crashes:

`>>> PROBLEM WITH THONNY'S BACK-END:

Traceback (most recent call last):
  File "D:\Progs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 340, in handle_command
    response = handler(cmd)
  File "D:\Progs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 864, in _cmd_upload
    written_bytes = self._upload_file(item["source"], item["target"], notify)
  File "D:\Progs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 1461, in _upload_file
    return self._write_file(block_generator(), target, notifier=notifier)
  File "D:\Progs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 1094, in _write_file
    result = self._write_file_via_serial(content_blocks, target_path, notifier)
  File "D:\Progs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 1133, in _write_file_via_serial
    raise RuntimeError("Problem opening file for writing: " + err)
RuntimeError: Problem opening file for writing: Traceback (most recent call last):

  File "<stdin>", line 4, in <module>

OSError: [Errno 19] ENODEV`

This seems similar to 
https://github.com/thonny/thonny/issues/1370
https://github.com/thonny/thonny/issues/1355
https://github.com/thonny/thonny/issues/936
etc.

Any suggestions?

aivarannamaa, 2020-10-25:
Do you have `/flash` subdirectory on you device? Did you try to upload your script on this directory?

jackho9, 2020-10-25:
Yes, but thonny crashes before that, as soon as I left click on 'Upload to /', after right clicking on the file I wish to upload.

If it helps, I can open an existing file in the M5StickC, say main.py, and copy-paste to it the contents of the file that failed to upload.  The updated main.py then saves and runs with no problems.

jackho9, 2020-10-25:
Ok, your question made me think of using the File menu and 'save as' in the M5StickC's /flash directory, which let me choose the /flash directory and it worked!

Perhaps when right clicking on the desired file in 'This Computer' the resulting menu should have the option of choosing the destination folder.  As it is currently it assumes that you want to send your file to the root directory ('Upload to /').  Is there a way to change the default directory that comes up?

Thanks for the pointer.

aivarannamaa, 2020-10-27:
> Perhaps when right clicking on the desired file in 'This Computer' the resulting menu should have the option of choosing the destination folder. As it is currently it assumes that you want to send [...]

The MicroPython part of the file browser is supposed to display the current directory (`os.getcwd()`) of the device. When I plug in my M5Stack, it is `/flash`. The context menu in the local part is supposed to offer upload to whatever directory is active in the upper part.

What does `os.getcwd()` say on your Stick after restart? Doesn't Thonny's file browser show the same directory?

You can change current directory by double-clicking a directory in the file browser or by clicking links in the header of the pane (to go upwards).

closed 2021-01-16 by aivarannamaa