thonny/thonny #1390: various issues when trying to work on code that uses tabs

iomintz, 2020-10-27:
These are small, separate, but related issues so I'll lump them into one GitHub issue for now. If you would like me to split them off into two issues, let me know.

- [ ] Tab to space conversion is prompted every time the user continues an indent
- [ ] Tabs are one character but treated as two indents semantically

## Tab to space conversion is prompted every time the user continues an indent

### Problem steps

1. Paste the following code which uses tabs:
```py
for x in range(3):
	print(x + 1)
    for y in range(3):
```
2. When prompted, choose not to replace tabs with spaces.
3. Press enter at the end of line 2.
4. Press enter on the last line (now line 4)

### Expected behavior

At step 3, `\n\t` is inserted is inserted. At step 4, `\n\t\t` is inserted.

### Actual behavior

- At step 3, the user is prompted to replace tabs with spaces. If they decide not to, `\n\t` is inserted.
- At step 4, the user is prompted to replace tabs with spaces. If they decide not to, `\n\t\t` is inserted.

### Remarks

This seems to be the editor trying to maintain consistency with the existing code, per the user's preference, while also forgetting that the user already decided not to convert tabs to spaces before in the same file. Suggested fix: recording somewhere (in memory or in a config file) that the user would prefer to keep tabs on paste.

While the intention is perhaps to discourage the use of tabs, this behavior comes off as heavy handed, to the point of actively being user-hostile. Tabs are discouraged by Thonny because the language discourages them, but Python also *requires* consistent indentation. Syntax takes precedence over style, so if it's not inconsistent behavior, it would seem to be inconsistent rationale.

## Tabs are one character but treated as two indents semantically

### Problem steps

1. Manually insert a tab character, using a clipboard or an IME.
2. When prompted, press "no" to not replace tabs with 4 spaces.
3. Press backspace on the tab character.
4. Note that the tab character is rendered as two indents, which is inconsistent with the behavior of replacing tab *keypresses* with one indent.
5. Save the file.

### Expected behavior

The entire tab character is deleted and the file is empty.

### Actual behavior

Step (2) inserts one tab character, but since it's rendered as two "indents", pressing backspace turns it into one indent, and the file contains one tab character.

### Remarks

Render a tab character as one indent (4 character widths / one tabstop). If you really want to distinguish them, display an indent marker, like this: 
[image]

## Notes

I don't want to start a flame war here =)

I respect the position of being opinionated for simplicity and for the sake of guiding beginners towards Good Practice and therefore not supporting tabs for indentation. However, I think it should be possible and not cumbersome for a beginner to contribute to codebases that already use tabs. If that use-case is out of scope for this project, feel free to close this issue. I am willing to try to fix these issues, if all contributors would rather not touch The Character That Shall Not Be Named.

aivarannamaa, 2020-10-27:
Actually I don't have strong feelings about tab. It was just easier to ignore them, because using tabs for indentation in Tkinter's Text widget is not really straightforward (#982). I've now learned that several users want or need to use tabs (#656, #505, #869, #599, #548).

I'd be really glad if you could sort out any of the issues mentioned here!

aivarannamaa, 2020-10-28:
I think the best solution would be making Thonny respect the original style when opening an existing file and offering a configuration option for new files. In this case the hated dialog can be removed altogether.

iomintz, 2020-10-28:
In that case, what would you do when pasting tabs into a file that uses spaces, or pasting tabs into an empty file?

aivarannamaa, 2020-10-29:
I think in the former case I would convert tabs to spaces and in the latter case I would put the file into tabs mode.

Do I understand correctly that you are willing to solve the tabs problem? If yes then do you think you can do it in 2 weeks? I hope to release stable Thonny 3.3 in about 2 weeks and this would be a nice addition to it.

iomintz, 2020-10-29:
If you can detail the subtasks that need to be done in order to close this issue, I may be able to provide a time estimate. As it stands I am unfamiliar with the codebase.

aivarannamaa, 2020-10-29:
Most important class is `thonny.tktextext.EnhancedText`. This whole module is actually independent of Thonny -- you can copy this module out of Thonny and create a simple Tkinter application for testing purposes.

As you can see, there are already some preparations done for supporting tabs -- look for `indent_with_tabs`, `replace_tabs`, `tabwidth` and `indentwidth`. The last two are taken over from IDLE code (they are used eg. in `thonny.roughparse` which is used with auto-indenting Python code and such). `indentwidth` tells specifies many spaces to use for indenting (if indenting with spaces) and `tabwidth` specifies the visual width of the tab character.

Python-specific and Thonny-specific additions are in `thonny.codeview`. Here the most relevant class is `SyntaxText`, which is actually used also for plain text (same editor can be switched between Python and plain-text mode by calling `set_file_type`). New editor starts in Python mode and can be switched to plain-text mode by saving the file with a non-Python extension.

I think the most import task is to change `thonny.tktextext.EnhancedText.check_convert_tabs_to_spaces` such that it doesn't create a messagebox anymore, but instead decides the need to replace the tabs from the existing content (and/or configuration variables).

Here are some things to test after your changes:

* Tab is also used for indenting selected lines (Shift-Tab fro dedenting), see `thonny.plugins.commenting_indenting`
* Tab in the end of the line is used for invoking code completion (subject to options Tools => Options => Editor).
* In plain text mode RETURN should carry over the indentation to the next line. In Python mode it is more complex (it may decrease on increase the indentation depending on what is on the current line)
* Backspace should remove entire indentation level even if it consists of 4 spaces.

aivarannamaa, 2020-11-10:
@iomintz, have you had chance to look at the code and provide an estimate? I don't mind doing it myself, it the code looks too messy for you.

iomintz, 2020-11-10:
Argh no I haven't

iomintz, 2020-11-10:
I don't have too much motivation to do it either, so if you are willing to do it I am happy to review

aivarannamaa, 2020-11-10:
I guess it's done now. I added comments about what I did under #599. Please reopen if you think there is something more to be done.

closed 2020-11-10 by aivarannamaa