thonny/thonny #1396: (Raspberry Pi) Window fails to gain focus on delegation

debugging-cymbal, 2020-10-31:
I am using Thonny on Raspberry Pi, and I have noticed that when I open a file from outside Thonny while it was opened, the main Thonny window gains focus for a very short amount of time, then the window I was opening from (for example, file manager or terminal) gains focus as well and overlaps the Thonny window, if they overlap.

My version of the system is Raspbian GNU/Linux 10 (buster).

### As I understand, this is how it should work:

1. You open a file from another application
2. Thonny notices there's already an instance running, so it exits
3. The main instance opens the file and becomes the topmost window (and stays the topmost window).

### What it does:

1. I open a file from another application (for example, file manager or terminal)
2. Thonny notices there's already an instance running, so it exits (When run from the terminal it says that it's delegating to another instance)
3. The main instance's window gains focus and immediately loses focus. (It's easier to notice if the windows overlap)

aivarannamaa, 2020-11-03:
Thank you for the report!

I can't reproduce this in Ubuntu. Will try Raspbian later.

aivarannamaa, 2021-01-16:
Is it still a problem with Thonny 3.3.2 ?

aivarannamaa, 2021-01-20:
I could reproduce this. Note to myself: `focus_force` in `became_active_window` seems to work in RPi. Needs testing on other platforms.

BionicPotato, 2021-01-21:
> Is it still a problem with Thonny 3.3.2 ?

It's still buggy on my computer with Thonny 3.3.3-dev. (the current state of the repo)