thonny/thonny #1401: OSX paste into file save dialog does not work

juanfal, 2020-11-03:
Last version (just installed and going like a charm), still with this annoying small problem.  I usually make new files in front of my students, you know, the header # myFile.py etc.  and since those names are sometimes a bit special, I always copy the file name from the very header to paste them into the (save ⌘-S) save dialog and then (⌘-V) paste the name.  Funnily, with Thonny (this a previous versions) it doesn't work.

The scrapbook is there as you can paste into whatever other place and application, but it doesn't paste into the save dialog

aivarannamaa, 2020-11-03:
Tkinter does have issues with clipboard (eg. https://bugs.python.org/issue40452)

Can you paste into Thonny's save dialog, when you copy something from another application?

juanfal, 2020-11-03:
Noup, no edition works.  Neither copying, nor pasting. They do not work neither using the menus, nor the keyboard, even when the flashing menu Edit works when you ⌘-[CXVZ], the do not work.  I’ve tried now even editing inside the dialog, trying to cut part of the edit box in the save dialog. Nothing.  The same when the clipboard comes from another application.

Thanks Aivar

-

aivarannamaa, 2021-02-27:
I could confirm this with https://github.com/thonny/thonny/releases/download/v3.3.5/thonny-3.3.5_alt.pkg (experimental Universal 2 build with Python 3.9.2).

It looks like Thonny captures the keyboard shortcut, because pasting via context menu works

juanfal, 2021-03-02:
I've tried 3.3.5 and it seems the problem doesn't go.  You can even use the menú items to Copy/paste, but neither works for the text in Open/Save dialogs

aivarannamaa, 2021-03-15:
A related issue: https://bugs.python.org/issue43496