thonny/thonny #1403: nicer/faster won't stop at breakpoints

DancingGuy, 2020-11-03:
This is somewhat like issue #1335 except for me no set-in-advance breakpoints are being honored. The screen shots prove it. This has happened 5+ times and I've tried numerous things on my own before reporting it here. It is a single script though it does use the eBay SDK for API calls to its database. The error is on eBay's side most likely, but I would like to be able to step the code so I can report it to them with details, and set up graceful error handling to continue the run in a redirected way via an except block. 
Using Thonny version 3.2.7 on Win 10 64-bit.
Thanks.
John
[https://drive.google.com/file/d/1VoWRq-ajKTjr4mxqiBWRa2zwDgzWzlHc/view?usp=sharing](url)

[https://drive.google.com/file/d/1ZyW3W5Q6-q86NiumRBkUrS15L6WsMBxu/view?usp=sharing](url)

[https://drive.google.com/file/d/11KwRLoZw_Ptzn8-xzGHBJTfLc-UREAIO/view?usp=sharing](url)

aivarannamaa, 2020-11-03:
Thank you for the report! 

I see that Thonny's exception serializer does something wrong. Someone has had similar problem: https://stackoverflow.com/questions/55838992/python-3-x-how-to-efficiently-split-an-array-of-objects-into-smaller-batch-fil

I'll look into this.

aivarannamaa, 2020-11-03:
Is one of the screenshot with the nicer debuger and the other with the faster one?

DancingGuy, 2020-11-03:
The screenshot with 4 breakpoints was with the faster debugger.  The other I don't recall. I run my script multiple times daily. Starting tomorrow I'll use the nicer ctrl F5 debugger exclusively until I can reproduce again the issue.

On Tuesday, November 3, 2020 Aivar Annamaa <reply@reply.github.com> wrote:

Is one of the screenshot with the nicer debuger and the other with the faster one?

DancingGuy, 2020-11-05:
I did a run with nicer Cntl-F5 and it behaved slightly better in that you can see that at least 2 user-supplied print statements were honored. Despite numerous breakpoints, however, the run did not stop. I also can not get Run-to-cursor to be available in the Run menu.
https://drive.google.com/file/d/1FjmEKVTgpvahXyNlZ-JDT1HEe0z0f0Q-/view?usp=sharing

aivarannamaa, 2020-11-07:
I could reproduce the internal error when debugging with Shift-F5. The fix will be in Thonny 3.3. Possibly this fixes your case with Ctrl+F5 as well. If not, then please reopen and post a short program, which demonstrates the problem.

closed 2020-11-07 by aivarannamaa