thonny/thonny #1405: MySql Connector not working

clarissa12193, 2020-11-07:
Whenever I make an Sql program on thonny and use the mysql connector, i get the error "Authentication plugin '{0}' is not supported".format(plugin_name))
mysql.connector.errors.NotSupportedError: Authentication plugin 'caching_sha2_password' is not supported

i would like to know how i can fix this asap

Thank You
[image]

aivarannamaa, 2020-11-07:
This doesn't look like Thonny's problem. Let me know if your code works outside of Thonny and doesn't work in Thonny.

closed 2020-11-07 by aivarannamaa

clarissa12193, 2020-11-07:
It works in other system's thonny, but I'm only getting this error in my
laptop

aivarannamaa, 2020-11-07:
You probably have configured MySQL differently in other systems. Try https://www.google.com/search?q=python+mysql+connector+authentication+plugin+%27caching_sha2_password%27+is+not+supported&rlz=1C1CHBD_enEE867EE867&oq=mysql+connector+python+auth&aqs=chrome.1.69i57j0i22i30i457j0i22i30l2.11537j0j7&sourceid=chrome&ie=UTF-8