thonny/thonny #1406: Vertical Scrollbar "Page up/down" doesn't work.

PeteCal, 2020-11-07:
With most vertical scroll bars, if you click above (or below) the indicator showing the area you are viewing, the screen jumps up (or down) one page to the next area.  Thonny (on my Raspberry Pi 3B+) doesn't do that.  Nothing happens.  

Other functions like Arrow Up (down) move it up (down) one line and the scrollbar can drag up and down.
No big thing but that is the way I habitually scroll.

aivarannamaa, 2020-11-08:
Thank you for the report! I was able to reproduce this with Raspberry Pi theme. Unfortunately I don't know how to keep the desired behavior while aiming for RPi-like appearance.

If for you this behavior is more important than the looks, then you could switch the UI theme to "Enhanced clam" (Tools => Options => Theme & Font).

PeteCal, 2020-11-08:
That is great!
Thanks for such a simple solution....very refreshing.

closed 2020-11-08 by aivarannamaa